bitcraze/crazyflie-firmware #604: unaligned memory access from packed structs warning in GCC 9.3

jpreiss, 2020-07-08:
Newer versions of GCC warn against taking the address of a member of a struct with `__attribute__((packed))`. I encountered this while building the Crazyswarm Python bindings for x86 with GCC 9.3. For example, if we run the following command from `src/modules`:

    gcc -c -Iinterface src/pptraj.c

we get warnings like

    src/pptraj.c: In function ‘poly4d_linear’:
    src/pptraj.c:199:16: warning: taking address of packed member of ‘struct poly4d’ may result in an unaligned pointer value [-Waddress-of-packed-member]
      199 |  polylinear(p.p[0], duration, p0.x, p1.x);
          |             ~~~^~~

At first, I thought this was a false positive because `struct poly4d` only contains `float` members, so even when packed they will always be aligned. However, looking in `mem.c`: https://github.com/bitcraze/crazyflie-firmware/blob/eff7ab9f83a7321a5221658326bda327aac7fc85/src/modules/src/mem.c#L567-L576
It does seem possible that the memory subsystem could end up placing the whole `struct poly4d` at an arbitrary alignment.

Do we care about this? I don't know a lot about the ARM architecture. The [Cortex-M4 Devices Generic User Guide](https://developer.arm.com/documentation/dui0553/b/) says that it supports unaligned access for the main load/store instructions (page 64 of pdf). But, then it says that unaligned accesses might be slow, so programmers should try to ensure aligned accesses. The list of unaligned load/stores does **not** include the special floating-point loads `VLDM` and `VLDR` (pp. 184-185 of pdf). Maybe the compiler will load the value to an integer register and then move it to a floating-point register?

If we are OK with unaligned accesses, then I guess no action is needed (aside from maybe disabling the warning if we move to a newer version of GCC).

If we want to avoid unaligned accesses, then I'm not sure what to do. I think these reads from `__attribute__((packed))` structs are common in this codebase.

krichardsson, 2020-07-27:
Yes, this is a problem in the later versions of GCC (also see #512) and the reason why we are still officially building with GCC8. We had a look before the summer but did not have the time to resolve the problem at that time.

I'm not sure but it might be that GCC8 aligns the struct (and packs the contents) while GCC9 does not align? We need to do some more digging into this to make sure the solution is OK. 

The current workaround is to compile with GCC8 (used by the toolbelt)

krichardsson, 2020-07-28:
-Wno-address-of-packed-member seems to turn the warning off. Remains to verify that the generated code is OK.

@jpreiss Did you see arbitrary alignment of the the start of the struct?

krichardsson, 2020-07-30:
The output from gcc 8 and 9 looks the same. As far as I can understand the difference is that they added the warning in gcc 9 but there is no difference in the generated code when it comes to packed structs. I assume it should be OK to disable the warning.

krichardsson, 2020-07-30:
I have disabled the warning to enable builds with gcc 9. We should investigate this further but that will be done in #512.

closed 2020-07-30 by krichardsson

ataffanel, 2020-08-01:
This is great!
To compile the Crazyflie firmware on a clean Ubuntu 20.04 the only requirement is now:
```
sudo apt install make gcc-arm-none-eabi
```

acxz, 2020-08-30:
I tried to get the latest master working but @ataffanel's solution did not work for me. I ran into the following error:
When running `make`
```
make[1]: arm-none-eabi-as: Command not found
make[1]: *** [tools/make/targets.mk:63: startup_stm32f40xx.o Error 127
```

See: #614

ataffanel, 2020-08-31:
This is surprising. I just tried in a docker image and the following command worked:
```
docker run --rm -it ubuntu:20.04
# In the docker shell ...
cd
apt install -y git make python3 gcc-arm-none-eabi
git clone https://github.com/bitcraze/crazyflie-firmware --recursive
cd crazyflie-firmware
make # build succeeds!
```