bitcraze/crazyflie-firmware #605: Expansion decks turned off for faster charging

knmcguire, 2020-07-13:
Some expansion decks draw a lot of power which will affect the charging time. Maybe it would be good to turn them off somehow with some define?

Based on the discussion going on here: https://forum.bitcraze.io/viewtopic.php?f=5&t=4311&p=19822#p19822

jonasdn, 2021-09-01:
This was discussed in the office, some notes from that discussion:

### Problem statements
- It is today not possible to turn of the expansion decks without also turning off the stm32 MCU at the same time
- The Crazyflie will charge its battery while turned off (the charging chip is active) but only at 100mA as opposed to 500mA
    - This is not indicated in anyway today
    - This is because to get 500mA you need to negotiate with the USB port and that is done by the nrf chip

### Possible solutions
1. We could send a message to the nrf chip from the stm32 asking it to shutdown the stm32 and expansion deck
    - This would need some plumbing to make possible for a user to do:
        - A parameter?
        - CRTP message?
    - We would still not have any indication of charge, should we:
        - Blink with the nrf led (M2) while charging?
        - Blink with  the nrf led in some pattern when message is received?
        - Do nothing?
        
 1.  Always do 500mA in off-mode, regardless of negotiation
     - So you could, as a user, turn off the Crazyflie using button while charging, to charge faster
     - Would still not have any indication, and nrf chip is off as well, so no means of LED flashing

1.  Have a special mode in nrf when cable is detected so it does not turn off, so it can keep charging at 500mA
    - Same charge indication problems as in 1.

Did I miss anything @krichardsson, @evoggy, @tobbeanton ?
Any thougts @knmcguire @ataffanel ?

jonasdn, 2021-10-05:
The way forward right now, we think, is option **3.** we will update the NRF firmware to always do 500mA charging, this will allow you to charge fast always, by turning the crazyflie off.

ataffanel, 2025-02-06:
The brushless makes this functionality even more important, we will give another look at it.