bitcraze/crazyflie-firmware #606: Error usisng micro-SD deck with flowdeck

knmcguire, 2020-07-27:
Reports of the initial test failing with the micro SD deck and the flowdeck:  https://forum.bitcraze.io/viewtopic.php?f=5&t=4380. The fix provided in https://forum.bitcraze.io/viewtopic.php?f=5&t=4186&p=19075&hilit=using+micro+SD+deck#p19075 (trying code with increased heap mem) does not seem to work. Also I have referred the user to issue #368 to add a delay in the microSD deck, but no cigar.

However, there might be a link with #368? Maybe the problem is in the ranging driver? Just a thought

wydmynd, 2020-08-16:
USD and flow together have a compatibility issue, I assume you are using a modified SD deck and the alternate spi flag?

knmcguire, 2020-08-18:
Actually, i noticed that there is already an issue about this open namely #270. There are some fixes in there described that can help you out, but to avoid duplications I will close this one.

closed 2020-08-18 by knmcguire