bitcraze/crazyflie-firmware #609: Add app layer examples to repository

knmcguire, 2020-08-13:
In all our other repos we usually have an example folder where we show scripts or code that uses the actual functionality of the repo. We don't have those for this firmware repo (also not for the NRF by the way), since that was a bit more difficult to do. 

However, since we have the app layer now since #502, it IS possible to add examples to this repository which can help out people that want to get certain things done with the onboard development. 

Here are some examples of what we can put in there:

- [ ] onboard push demo (currently only available on this repo: https://github.com/ataffanel/crazyflie-push-demo)
- [ ] P2P helloworld ( this documentation does not seem to be enough for people who start working on it: https://www.bitcraze.io/documentation/repository/crazyflie-firmware/2020.02/p2p_api/. We usuallly refer people to this [repo](https://github.com/tudelft/SGBA_cf2_app_layer) but that is usually a bit too complex )

Maybe some other ideas we can think off,

tomsoferRND, 2020-08-16:
I will add my "flying without crashing" based on push demo, and slight modifications to commander.c
would like to see examples that use the log and parameter frameworks, especially the ability to change parameters from within the code.

knmcguire, 2020-09-01:
It should be almost ready. I just want to add a security check to the push demo though, so that it does not take off if it the multiranger or flowdeck are not detected

knmcguire, 2020-09-02:
Alrighty then. I will close this issue now. 3 examples is good enough to start with :) :)

closed 2020-09-02 by knmcguire