bitcraze/crazyflie-firmware #610: Investigate possibility to implement BLHeli pass though interface

tobbeanton, 2020-08-17:
The BLHeli pass though interface is an convenient way to configure/flash BLHeli ESCs "via" the flight controllers over USB. Investigate the possibility to implement this on the Bolt. Hopefully the open source drivers on beta/cleanflight can adapted and used.

tobbeanton, 2020-10-22:
For the Bolt this does not seem possible withhout doing HW modifications. One of the motor outputs lacks a DMA channel for the timer. The BigQuad deck might still be possible though and requires more investigations.

jonasdn, 2021-09-03:
Is this something we still want to do? Do we have a plan we can specify here?

jonasdn, 2021-10-04:
Not possible for current hw platforms.

qubitter, 2021-11-05:
Would this be doable for a 4-in-1 ESC, given that it receives connections from all 4 motors? Given that the HW for the bolt is finalized, is this still not doable? BLHeli passthrough is a fairly widespread feature in most modern flight controllers.

tobbeanton, 2021-11-05:
After further investigations a pass though is possible but not DSHOT. This functionality should be implemented as part of getting Bolt out of early access.