bitcraze/crazyflie-firmware #611: Pre-arm ESCs

Blake-alt-del, 2020-08-18:
Is it possible to make a prearm state for brushless motors? I am wanting to use the crazyflie-firmware to arm and spin up brushless motors to a low threshold (around 5000) right after the battery is plugged in.

I explored the idea of creating a 'prearm' state where the motors spin up to around 5000 on startup. I found that motor values can be edited here: https://github.com/bitcraze/crazyflie-f ... #L100-L110
However, motorSetEnable needs to be set to 1 in order for the motors to actually spin. Where would be the best place to do this? I tried in power_distribution_stock.c but that interrupted the startup sequence where the motors beep as they are tested, and the drone won't fly.

It seems as though the ESC won't arm because it want's "0" power sent to it first. Is there any way around this?

tobbeanton, 2020-09-01:
I think a simple state machine is needed for this. As you say, the ESC want's "0" power to calibrate. Therefor I suggest we add a armed state for brushless setups. This could be triggered either by setting a parameter though python API or by e.g. holding the controller sticks in some predefined way.

As a quick hack I think you could do something like the below in the power distribution function.
`
#define MIN_THRUST 5000
...
void powerDistribution(const control_t *control)
{
  control->thrust = fmaxf(control->thrust, MIN_THRUST);
...
`
This should spin up the motors to your set MIN_THRUST level as soon as the sensors are calibrated.

Blake-alt-del, 2020-09-19:
I like the quick hack idea, I tried it and it wouldn't build as it looks like thrust is a read-only object. quite odd. 

No doubt it would be neat to set a parameter in Python API. I am using ROS to run the crazyflie client and sending high level commands directly to my drone, however I haven't figured out how to write directly to the drone motors yet. I was hoping the prearm could be written into firmware so that it happens after the motor test.

closed 2021-08-30 by jonasdn