bitcraze/crazyflie-firmware #614: Compilation on Ubuntu 20.04 Fails arm-none-eabi-as not found

acxz, 2020-08-30:
I ran into the following error when running `make`
```
make[1]: arm-none-eabi-as: Command not found
make[1]: *** [tools/make/targets.mk:63: startup_stm32f40xx.o Error 127
```

acxz, 2020-08-30:
The following solution worked for me: https://askubuntu.com/a/1243405

ataffanel, 2020-08-31:
How did you install gcc-arm-none-eabi? As noted in my answer to your [other message](https://github.com/bitcraze/crazyflie-firmware/issues/604#issuecomment-683583807) it should be enough to install the gcc-arm-none-eabi package from ubuntu 20.04 to build the Crazyflie firmware (plus make, git and python3).

Did you start from a system upgraded from 18.04 or did you have other PPA activated that could have interfered?

acxz, 2020-08-31:
> How did you install gcc-arm-none-eabi?

`sudo apt install gcc-arm-none-eabi`

> Did you start from a system upgraded from 18.04 or did you have other PPA activated that could have interfered?

I did start from a system with 18.04, but had no PPAs that could have interfered.

Based on the list of files in Ubuntu 20.04's `gcc-arm-none-eabi` package (https://packages.ubuntu.com/focal/amd64/gcc-arm-none-eabi/filelist), the program `arm-none-eabi-as` does not exist.

acxz, 2020-08-31:
Can you post the output of  ``dpkg -S `which arm-none-eabi-as` ``?

Maybe that will give us some insight.

ataffanel, 2020-08-31:
> Can you post the output of `` dpkg -S `which arm-none-eabi-as` ``?

```binutils-arm-none-eabi: /usr/bin/arm-none-eabi-as```

> Based on the list of files in Ubuntu 20.04's `gcc-arm-none-eabi` package (https://packages.ubuntu.com/focal/amd64/gcc-arm-none-eabi/filelist), the program `arm-none-eabi-as` does not exist.

```gcc-arm-none-eabi``` depends on ```binutils-arm-none-eabi``` which does contains ```arm-none-eabi-as```: https://packages.ubuntu.com/focal/amd64/binutils-arm-none-eabi/filelist.

acxz, 2020-09-01:
hmm weird, reinstalling `gcc-arm-none-eabi` worked.

Thanks for the quick responses!

closed 2020-09-01 by acxz