bitcraze/crazyflie-firmware #615: Charging with Qi-deck might trigger charge temp check that slows down charging.

tobbeanton, 2020-09-03:
As first temp check was implemented in 2020.02 and updated/improved in 2020.04 it can still trigger the charge temp threshold, with the Qi deck, which will disable charging until temp is below again. This is to stay within the 45 deg charge temp specification of the battery. The Crazyflie will still be charged but slower do to the type of trickle charging that happens.

Workarounds:
1. Disable temp check by building the nrf firmware with DISABLE_CHARGE_TEMP_CONTROL defined. Might wear out battery faster.
2. Turn on one or more of the motors slowly to work as a fan while charging

jonasdn, 2021-09-03:
So, should we implement a fix/workaround for this? If so which one? What is the plan?

jonasdn, 2021-10-04:
This is similar to #605 (Expansion decks turned off for faster charging).

jonasdn, 2021-10-05:
We will attempt to solve this by better documenting better this over-temp state, so as to not alarm users.

The turning of motors is a fun and cool feature, we will not work on it at the moment, but if anyone wants to do it:

- It should be a compile-time option to enable
- It should warn that it is only, right now, makes sense for the Crazyflie platform