bitcraze/crazyflie-firmware #617: Kalman estimator diverges when using flow deck on dark floor

MauroPfister, 2020-09-11:
Hi

I am flying a crazyflie 2.1 (latest firmware) with a flow deck v2 over a floor covered with black-white textured rubber mats (see attached picture). When the drone is sitting on a black spot, the x and y position estimates start diverging immediately. On the white parts of the rubber mats, the estimation is stable. I attached a screenshot showing this phenomena in the `cfclient`. The diverging position estimation often causes the crazyflie to crash on take off when flying with the hover assist mode.

[image]

[image]

I understand that the flow deck might not work well on a dark surface but I thought that with the improvements from #608, the flow information would not be used while the crazyflie is sitting on the ground? Are there any parameters I could tune or is this maybe a bug in the firmware?

Thanks!

krichardsson, 2020-09-17:
Hi and thanks for your report!

One of the changes in #608 was to only use the motion data from the flow sensor when it also reports motion (a different parameter). It seems as it is sometimes reporting motion also when the CF is sitting on the floor and this is why you get the results you see.

We have noticed that the flow data is pretty unreliable at short distances and it might be a good idea to add a distance check as well to not use the flow data unless the distance sensor reports a distance larger than some min distance (maybe 20-30 mm?) One complication would be that on the IMU would be the only input for the kalman estimator on the ground, and the estimate would drift away quickly. It could be possible to add some logic to push zero motion into the kalman filter to remedy this, but there is a risk it will interfere with other decks (for instance using flow + LPS).

A fairly correct solution that can be applied on the current firmware is to reset the estimator just before takeoff.

MauroPfister, 2020-09-17:
Thanks for the detailed answer @krichardsson! 

How would I make sure that the estimator is reset just before takeoff? I tried setting the parameter `kalman/resetEstimation` to `1` but it did not seem to make much of a difference. When I leave the CF on a black spot for a few seconds until the estimation diverges and then take off, the CF still drifts considerably before stabilizing in the air.

krichardsson, 2020-09-17:
Yes, setting kalman/resetEstimation is the way to do it.

If you are using the MotionCommander class I see that we already reset the estimator when taking off (https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/positioning/motion_commander.py#L95). There is a 2 seconds sleep (https://github.com/bitcraze/crazyflie-lib-python/blob/161583f01d940694f1aa47ced0fd49c9b915598b/cflib/positioning/motion_commander.py#L412) though that I think might be a bit too long. You could try to remove it.

MauroPfister, 2020-09-17:
I have not used `crazyflie-lib-python` yet. All my experiments were done using either `cfclient` or the `crazyswarm` package. 
So what exactly happens when I set `kalman/resetEstimation` to `1`? Does it reset the kalman filter once and if I want to reset it again, I have to switch the value from 1 to 0 and then back to 1? Or does it mean that every time that I take off, the kalman filter gets reset?

krichardsson, 2020-09-22:
crazyflie-lib-python is used by the client and was installed with the client

When you set kalman/resetEstimation to 1, the kalman filter is reset to an initial state (some hard coded, some parameterized). You actually don't have to set it back to 0, a second write of a 1 will reset the filter again. The reset happens when you set it to 1, not at take off.

MauroPfister, 2020-09-25:
Thanks for the clarification! From my side the issue can be closed.

krichardsson, 2020-09-28:
Great!

closed 2020-09-28 by krichardsson