bitcraze/crazyflie-firmware #620: Unable to Clone - Submodule FreeRTOS: reference is not a tree

Yaxit, 2020-09-15:
Hello,
I'm just starting with crazyflie and I'm having issues in cloning the firmware.
I run `git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git` as described in the readme file, but the process fails when cloning in vendor/FreeRTOS with error `Unable to checkout '82e80521124c6e96e5fdb3538a2533f994e2db8f' in submodule path 'vendor/FreeRTOS'`

I then tried to run git submodule init and git submodule update and I get this details:
`fatal: reference is not a tree: 82e80521124c6e96e5fdb3538a2533f994e2db8f
Unable to checkout '82e80521124c6e96e5fdb3538a2533f994e2db8f' in submodule path 'vendor/FreeRTOS'`

I read that this might be a misconfiguration of the .gitmodules file but I'm not sure I understand how to fix that.
Can you help me?

krichardsson, 2020-09-15:
Hi!

Not sure why you get this problem, it works on my machine. When I try it I get

```
MacBook-Pro:tmp kristoffer$ git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git
Cloning into 'crazyflie-firmware'...
remote: Enumerating objects: 36, done.
remote: Counting objects: 100% (36/36), done.
remote: Compressing objects: 100% (30/30), done.
remote: Total 14947 (delta 8), reused 17 (delta 5), pack-reused 14911
Receiving objects: 100% (14947/14947), 13.29 MiB | 4.95 MiB/s, done.
Resolving deltas: 100% (9482/9482), done.
Submodule 'vendor/CMSIS' (https://github.com/ARM-software/CMSIS) registered for path 'vendor/CMSIS'
Submodule 'vendor/FreeRTOS' (https://github.com/FreeRTOS/FreeRTOS-Kernel) registered for path 'vendor/FreeRTOS'
Submodule 'vendor/cmock' (https://github.com/throwtheswitch/cmock.git) registered for path 'vendor/cmock'
...
Cloning into '/Users/kristoffer/code/bitcraze/tmp/crazyflie-firmware/vendor/FreeRTOS'...
remote: Enumerating objects: 56, done.        
remote: Counting objects: 100% (56/56), done.        
remote: Compressing objects: 100% (20/20), done.        
remote: Total 149399 (delta 38), reused 36 (delta 36), pack-reused 149343        
Receiving objects: 100% (149399/149399), 103.56 MiB | 5.22 MiB/s, done.
Resolving deltas: 100% (108138/108138), done.
...
```

Maybe it is a temporary problem? Can it be related the your git installation?

SuperN1ck, 2020-09-17:
Hi!
We also have the same problem since a couple of days.
We resolved it by ignoring the error and manually performing a pull
```
cd vendor/FreeRTOS
git fetch --all
git checkout master
git pull
```
But since there is a specific branch defined in the `.gitmodules`-file I'm not sure if checking out master is recommended.

Yaxit, 2020-09-18:
Hi,
It was indeed related to the git version. I was running Ubuntu 16 with git 1.6 I think.
I now installed Ubuntu 18 with git 1.17 and the problem disappeared.
Still an interesting problem though!

krichardsson, 2020-09-18:
I have no idea why it does not work. 

A basic test is to clone using a docker container
```docker run --rm alpine/git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git```

The latest version of FreeRTOS should be fine, I actually just bumped to the latest version in #621.

krichardsson, 2020-09-18:
@Yaxit Great that it works and thanks for the feedback!
@SuperN1ck Might be worth checking your git version

SuperN1ck, 2020-09-18:
@krichardsson we are running git 2.7.4 on Ubuntu 16.04 but I just checked again and the issue seems to be resolved now (by bumping to the latest version I guess?). Thank you!

krichardsson, 2020-09-18:
Great!

closed 2020-09-18 by krichardsson