bitcraze/crazyflie-firmware #631: Add functionality to push CRTP packets to the firmware, for instance from an app

krichardsson, 2020-10-19:
Add functionality to push CRTP packets to the RX queue from within the firmware. This would make it possible for an app to send set points or other data to the system. 
Even better would be to replicate (most of) the low level API in the python lib.

See also #502

ataffanel, 2020-10-20:
Note that the protocol is designed to have only one client connected. If CRTP packets comes from an app in the Crazyflie, a client cannot be connected via the radio. In that case I suggest using the link switch we have already in place for radio/usb/serial and add an internal link to it.

krichardsson, 2020-10-20:
Good point!
The end goal (at this time) is to mirror some functionality of the python lib in the firmware to make it easier to port python apps to firmware apps. CRTP is not the way to go, a better solution is to add a public API on a module level for now. 
If we want to add other other interfaces in the future (for instance a REST API), they will not use CRTP so it would be the wrong solution now.

closed 2020-10-20 by krichardsson