bitcraze/crazyflie-firmware #635: assert kalman with lighthouse

knmcguire, 2020-10-26:
Recently we have had several reports about an assert in the kalman filter when using the lighthouse system. 

`SYS: Assert failed at .//src/modules/src/kalman_core.c:253`

This was reported in this forum thread: https://forum.bitcraze.io/viewtopic.php?f=20&t=4456

krichardsson, 2020-10-28:
This is most likely related to LH2. It seems as we sometimes push angles that are 0.0 for all sensors into the estimator, maybe this can case this problem? It is not unlikely that it will cause problems in the (hackish) yaw estimation.

We should obviously catch this case before we use the data, I guess the problem is somewhere in the pulse_processor_v2.c file.

krichardsson, 2020-10-29:
The 0.0 angles seems to be the output of the first sweep that is received, but this is not the cause of the assert. This is caused by a divide by zero here: https://github.com/bitcraze/crazyflie-firmware/blob/dbefcf6281b4c22f32316b2c9a231cc0a7d80b22/src/modules/src/kalman_core.c#L613

closed 2020-10-29 by krichardsson