bitcraze/crazyflie-firmware #653: hardcoding basestation geometry is deleted somewhere after

knmcguire, 2020-11-19:
People have been reporting the kalmanfilters position diverting after hardcoding the basestation geometry. See here the forum: https://forum.bitcraze.io/viewtopic.php?f=20&t=4537. 

Since this is an instruction of the lighthouse setup, this will need to be fixed quickly or at least provide an alternative.

https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/lighthouse/setting_up/

krichardsson, 2020-11-20:
The lighthouse code is under heavy development and it is possible that the latest version of the master branch will be unstable for a few weeks.

For a stable code base we suggest the following versions:
* The 2020.09 tag is the latest official release
* The 463110b1aae6166fdecdcfca4d3bc64532e949bc commit should essentially contain what was used in the tutorial and should also be stable.

krichardsson, 2020-11-20:
The root cause of this problem is that the geo and calib data has been moved to the lighthouseCoreState. lighthouseCoreState is located in the CCM memory that is always initialized to zero (regardless of any default values in the code) and this is why the values are not set. 

Some of the ongoing work is related to provide functionality to set/get and persist this data, this is the long term solution. 

Two short term workarounds are:
* Create an app that uses the ```lighthousePositionSetGeometryData()``` and ```lighthouseCoreSetCalibrationData()``` functions to set the desired values
* Remove the ```NO_DMA_CCM_SAFE_ZERO_INIT``` on this line https://github.com/bitcraze/crazyflie-firmware/blob/8c850fd0cfa3fab81dabbef1b865348eb716a98f/src/modules/src/lighthouse/lighthouse_core.c#L79 which will move the data to normal RAM and initialize it the normal way.

knmcguire, 2020-11-20:
I made a gist for an app layer implementation of setting the geo and calib data.

https://gist.github.com/knmcguire/040df041b9b9e1c90662e20f165d85a5