bitcraze/crazyflie-firmware #675: CFloader not accessible through snap

knmcguire, 2021-01-25:
On Ubuntu 20.04, If the crazyflie firmware is build and needs to be uploaded by 

`make cload`

then the error message appears if the latest release of the cfclient is installed by snap

```
python3 -m cfloader  flash  cf2.bin stm32-fw
/usr/bin/python3: No module named cfloader
make: *** [Makefile:415: cload] Error 1
```

This does not happen if the released cfclient is installed by pip3

ataffanel, 2021-01-27:
The snap is completely isolated from the local python installation so this command will not work when the client is installed using the snap.

There is a couple of things we could do:
 - Document that the snap cannot be used to run the bootloader, only to run the client GUI. This means documenting that ```make cload``` requires ```pip install cfclient```. I think this is what we should start with.
 - Make the bootloader accessible in the snap. The command would change in the Makefile but we could make it automatically detect what command to use (or even have the bootloader as a command-line argument to the client).
 - Move the bootloader to another pip package so that it can be installed independently from the client GUI.

As a side note: the snap and the local python environment are separate, this means that there is no problem installing both at the same time, which gives the advantage of the snap being automatically updated and available in the system menu, and the client installed via pip giving access to the bootloader.

jonasdn, 2021-10-04:
The snap is meant to expose the cfclient only, similar to the exe file for windows, supporting cfloader is out-of-scope.

closed 2021-10-05 by jonasdn