bitcraze/crazyflie-firmware #679: Lighthouse status reporting

krichardsson, 2021-02-03:
Make it easier for the user to understand the current status of the lighthouse system by adding reporting. Reporting should be done by logging to be displayed in the client, but also by using the LEDs on the lighthouse deck.

cafeciaojoe, 2021-02-06:
I have tried logging validAngles and this causes a hard fault on the cf. See thread https://forum.bitcraze.io/viewtopic.php?f=16&t=4687&p=21402#p21402

krichardsson, 2021-02-08:
@cafeciaojoe The validAngles functionality was contributed by a user for LH V1 and I don't know what the current status is, but it has not been implemented for LH V2 yet. Obviously it should not crash though, please open a new issue for it, as this issue is mainly intended for the new functionality we are adding.

closed 2021-02-24 by krichardsson