bitcraze/crazyflie-firmware #681: Safelink might lose packets

whoenig, 2021-02-05:
The current safelink implementation has several potential places in the firmware, where packets can be lost if an internal queue is full, because return codes are not always checked or blocking calls are not used. This issue is for tracking changes that improve this handling in the STM firmware.

ataffanel, 2021-02-08:
Thanks for the issue and the PRs! I renamed the title and ticket to reflect that packets are lost in the CRTP stack and not in safelink (safelink only protects the radio packets, and so far it works).

jonasdn, 2021-03-18:
Is this still an active issue? @ataffanel @whoenig ?

whoenig, 2021-03-18:
We can close this. The remaining work/bug is captured in #703.

closed 2021-03-18 by whoenig