bitcraze/crazyflie-firmware #689: uSD card not able to logging when using lps-deck

easy98, 2021-02-07:
Hi,

When I using lps-deck, the uSD card is not able to log any variables. I find this problem has been solved in this issue #270 . But I am still a little bit confused. Do I still need to patch and solder the uSD card deck or I only need to modify some codes to solve this problem?

Thanks in advance.

tobbeanton, 2021-02-09:
What FW version are you running? I tested this on the latest master branch and could not find any problems. There are some debug logging variables you could check, see image below. There you can see the write actions on the SPI bus and to the SD card.
[image]

Is logging working without the lps-deck? Perhaps you are logging to much. There is a limit on how much can be logged as the SPI bus is shared and if positioning is in TWR mode this occupies the SPI bus with a lot of transactions.

easy98, 2021-02-10:
Thank for reply!

Logging works well without lps-deck. When I add lps-deck, the CF will restart after a few seconds, and the lps-deck will not blink.  It doesn't work even there is only one logging variable. I use the latest released firmware.

krichardsson, 2021-03-01:
Hi!

Can we close this issue?

easy98, 2021-03-01:
Sorry for the delay, you can close this issue. Thanks!

closed 2021-03-01 by krichardsson

easy98, 2021-06-18:
Hi, 
How can I check the logging variables as the images shown to make sure the uSD work correctly? Thanks!