bitcraze/crazyflie-firmware #690: Questions about log parameters

864419409, 2021-02-07:
Hello, I recently studied how to record parameters by daily logging, but I was not quite sure about the physical meaning of some parameters in the process of choosing parameters.I inquired the forum and many places all did not find the complete explanation document, would like to ask, where should I inquire their physical meaning?The following figure shows some of the parameters that might be used.
Thank you for your answer.
[image]

krichardsson, 2021-02-08:
Unfortunately there is currently no documentation for the log variables, the best place for information is the source code. You can find them by searching the code base for "LOG_GROUP_START" which will tell you where the log groups are defined. In those group blocks you can find the individual log variables, mainly defined using the "LOG_ADD" macro. The second argument of the LOG_ADD macro is the name and the last argument is the actual C-variable that is logged. You can search for where this variable is used to figure out what it means.

864419409, 2021-02-09:
Thank you for your reply. I will try your method right away. If it is convenient, could you please tell me where to extract the source code of this piece?

krichardsson, 2021-03-01:
Hi! 

Can we close this issue?

864419409, 2021-03-07:
yes of course，  Sorry, I'm still learning how to use the website

closed 2021-03-08 by krichardsson