bitcraze/crazyflie-firmware #695: If voltage goes below ~2.6V it can reset the LH-deck

tobbeanton, 2021-02-11:
For a longer discussion check [this](https://forum.bitcraze.io/viewtopic.php?f=20&t=4679&p=21423#p21423) forum thread.

Summary: Especially during take off and if old batteries are used the LH-deck can be reset if voltage goes below ~2.6V, resulting in lost tracking and most often a crash.

One solution could be to reduce output power if voltage goes below e.g. 2.7V. This is specific to each platform so implementation must be platform dependent.

jonasdn, 2021-09-03:
What does platform mean in this context @tobbeanton ? Different decks or Crazyflie 2.1 vs BigQuad vs Tag vs Quad? or both?

jonasdn, 2021-09-03:
Would having a voltage limit under an #ifdef be a sufficient solution for this?

closed 2026-10-01 by ArisMorgens