bitcraze/crazyflie-firmware #700: Flash lighthouse deck bitstream over CRTP

ataffanel, 2021-02-18:
We currently embed the lighthouse deck bitstream in the Crazyflie firmware. This allows for easily upgrade the bitstream and making sure the Crazyflie firmware is kept in sync with new version of the bitstream. However, this is causing the Crazyflie firmware to be ~128KB bigger and this technique will not scale if we have more deck that requires a firmware or bitstream.

The solution is to be able to flash the bitstream over CRTP, either using the radio or USB.

The current plan is to:

- [ ] Check that the deck contains a supported bitstream by calculating and comparing the bistream CRC at startup
- [ ] Making the deck flash available to read and write from the ground by making it accessible as a memory in the memory subsystem
- [ ] Expose information about the deck memory as parameters/logs to allow the lib to discover what firmware is needed. Ideally this is done in a namespace that will allow future deck to expose firmware requirement in a similar way.

krichardsson, 2021-03-11:
Wait to merge this until we can make a release of firmware in the new format

knmcguire, 2021-03-18:
It is now resolved with PR  #736

closed 2021-03-18 by knmcguire