bitcraze/crazyflie-firmware #704: uSD Deck: Add event logging support

whoenig, 2021-02-23:
The current usd deck driver supports logging to uSD card at a fixed frequency. However, this does not allow to log asynchronous events, such as ones that occur when a new sensor measurement is available. Ideally, we support both event logging and regular fixed-frequency logging.

# Proposed API

Similar to params, logs, and decks, a module can define an event datatype:

New data structure:
```
typedef struct usddeck_eventtype {
  uint16_t event_id;
  uint16_t num_variables;
  char* typestring;
} usddeckEventType;

```
new macro:
`USDDECK_EVENT()`

A module can then call the following function to add a new event, which will be automatically timestamped using the usec-Timer:

```
usddeckAddEvent(const usddeckEventType*, const uint8_t* data)
```
Here, the size is automatically known based on the event type, because events are assumed to be binary encoded & fixed size.

# Proposed File Format

For reference, the current file format is as follows:

```
uint8_t  num_log_variables
varname1(vartype1),varname2(vartype2),...,varnameN(vartypeN),
<CRC>
uint8_t sets_to_write; binary data for all sets; <CRC>
```

The new, proposed file format generalizes this concept by adding an event type to each dataset and the beginning of the header can list variable names and types for each event type:

```
uint8_t 0xBC header
uint16_t version
uint16_t num_event_types
for each event type:
   uint16_t event_id
   uint16_t num_variables
   varname1(vartype1),varname2(vartype2),...,varnameN(vartypeN),
for each event:
   uint16_t event_id
   uint32_t timestamp (usec timer)
   data (length defined by TOC event type)
```

# Notes

* This will be a breaking change in terms of file format. However, the new file format is finally versioned, and both decoding scripts and usddeck driver are in the same repository, so this should be OK. Adding the magic 0xBC header at the top, should allow us to detect the new format with high probability.
* There is currently no CRC in this design; an invalid record can be easily detected by looking at the file length. This is because I've never observed a CRC check failure. Alternatively, we can also CRC over larger chunks or over the whole file.
* We should avoid closing/opening the file frequently, as we currently do, as this creates most of the SPI traffic. This likely means that we are less resilient to firmware crashes compared to before. However, debugging a firmware crash does not seem to be a good use case of the uSD card deck in the first place?

krichardsson, 2021-02-23:
Ping @Williamwenda

tobbeanton, 2021-02-23:
I think this is a great start.

whoenig, 2021-03-13:
This has been merged.

closed 2021-03-13 by whoenig