bitcraze/crazyflie-firmware #711: Unified estimator task loop

knmcguire, 2021-03-02:
Currently the extended Kalman filter is always initialized and taking up RAM, even if only the complementary filter is used. It is not really adding problems now, but if in the future more types of estimators are added, this might be a bit difficult to handle.

How about just starting one generic estimator task, where within we can switch between the Comp. filter, EKF or estimator number# ?

jonasdn, 2021-09-03:
Is this something we want to pursue? If so, do we have a plan of attack?

knmcguire, 2021-09-06:
To kind of give a background information: In the current state of the firmware, the kalman estimator is already initialized [here](https://github.com/bitcraze/crazyflie-firmware/blob/d7d9c3dfbe1dd13fe0376be5632281a74a0c968a/src/modules/src/system.c#L190) in system.c and the rest of the estimators [here in stabilizer.c ](https://github.com/bitcraze/crazyflie-firmware/blob/d7d9c3dfbe1dd13fe0376be5632281a74a0c968a/src/modules/src/stabilizer.c#L171). The kalman filter is currently now setup as a seperate task, but I think we should also do that with the complementary and the OOT one.

Technically, the kalman filter is not taking up RAM if the complementary filter is selected, as the loop is physically stopped [here](https://github.com/bitcraze/crazyflie-firmware/blob/d7d9c3dfbe1dd13fe0376be5632281a74a0c968a/src/modules/src/estimator_kalman.c#L207) if the complementary filter is selected, so that was a wrong assumption from me, but... the general structure could be improved, starting with putting the complementary filter also in a task for starters

knmcguire, 2022-03-11:
This would also be valid if we want to make python bindings of these api's like mentioned in #602

krichardsson, 2023-08-22:
There is some documentation in the oot build section https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/oot/