bitcraze/crazyflie-firmware #713: FatFS: upgrade to latest version and move to vendor folder

whoenig, 2021-03-02:
We should upgrade to the latest FatFS version and at the same time clearly mark it as an external dependency. For that, we could try to find (or create) a github mirror of FatFS, or put it directly in a subfolder in vendor/.

This is needed for clang support: https://github.com/bitcraze/crazyflie-firmware/pull/708.

ataffanel, 2021-03-04:
What about using this repos that seems to be mirroring the FatFS versions at regular intervals: https://github.com/abbrev/fatfs?

whoenig, 2021-03-05:
I tried that, and I think it doesn't work well because it comes with an example ffconf.h (https://github.com/abbrev/fatfs/tree/master/source). The [way the config is included](https://github.com/abbrev/fatfs/blob/master/source/ff.h#L29) makes it impossible to deal with this cleanly with include paths. So I am afraid we might need to stick with a copy. The question is just the location. Since it is external code, should it still go into vendor? I am unsure about the difference between src/lib/ and vendor/ folders.

ataffanel, 2021-03-07:
```src/lib/``` is historical and is for external copy-pasted libs. ```vendor/``` was created with the first submodule and is used only for submodule.

So if we want to follow this convention we should keep the FatFS copy in ```src/lib/```. I feel it makes some sense to keep both kind separated since there is a very strong guarantee for a submudule that we have not modified it, while we apparently have tweaked some of the copied lib, so having them in ```src/lib/``` indicates that the lib might be a crazyflie-tweaked version.

closed 2021-03-17 by whoenig