bitcraze/crazyflie-firmware #729: Question about Multi-ranger Driver and Ranging Frequency

hanzhi713, 2021-03-12:
Hi, in my application, I need high-frequency ranging data from the multi-ranger deck. The current multi-ranger deck driver https://github.com/bitcraze/crazyflie-firmware/blob/347e630aec4a283384e22b7c0588123d66f4c479/src/deck/drivers/src/multiranger.c#L108 seems to only update ranging data at 10Hz. On the other hand, the zranger which uses the same sensor updates the range at 40Hz. https://github.com/bitcraze/crazyflie-firmware/blob/347e630aec4a283384e22b7c0588123d66f4c479/src/deck/drivers/src/zranger2.c#L132

Is there a specific reason to make their frequencies different?

hanzhi713, 2021-03-12:
Also, I need to modify the driver to change the ROI. The code snippet below shows my modified version of the multi-ranger driver. While working on it, I have a couple of questions:

1. Why are sensor init calls placed in mrTest and not mrInit? https://github.com/bitcraze/crazyflie-firmware/blob/347e630aec4a283384e22b7c0588123d66f4c479/src/deck/drivers/src/multiranger.c#L155
2. The documentation of vl53l1x suggests that VL53L1_ClearInterruptAndStartMeasurement can be used to initiate the next measurement. However, `VL53L1_StopMeasurement` and `VL53L1_StartMeasurement` are used when the next measurement is started. My testing showed that `VL53L1_ClearInterruptAndStartMeasurement` indeed does not work, but why is that?

```c
static uint16_t mrGetMeasurementAndRestart(VL53L1_Dev_t *dev) {
    VL53L1_Error status = VL53L1_ERROR_NONE;
    VL53L1_RangingMeasurementData_t rangingData;
    uint8_t dataReady = 0;
    uint16_t range;

    status = VL53L1_GetMeasurementDataReady(dev, &dataReady);
    while (dataReady == 0) {
        vTaskDelay(M2T(1));
        status = VL53L1_GetMeasurementDataReady(dev, &dataReady);
    }

    status = VL53L1_GetRangingMeasurementData(dev, &rangingData);
    range = rangingData.RangeMilliMeter;

    // why is this function not working?
    // status = VL53L1_ClearInterruptAndStartMeasurement(dev);
    status = VL53L1_StopMeasurement(dev);
    status = VL53L1_StartMeasurement(dev);
    status = status;
    return range;
}

static bool setupSensor(VL53L1_DEV dev, I2C_Dev *I2Cx) {
    if (!vl53l1xInit(dev, I2Cx))
        return false;

    VL53L1_Error status = VL53L1_StopMeasurement(dev);

    // status += VL53L1_SetDistanceMode(dev, VL53L1_DISTANCEMODE_MEDIUM);
    status += VL53L1_SetMeasurementTimingBudgetMicroSeconds(dev, 25000);
    status += VL53L1_SetInterMeasurementPeriodMilliSeconds(dev, 30);

    // we can configure ROI (FOV) here
    VL53L1_UserRoi_t roiConfig;
    roiConfig.TopLeftX = 6;
    roiConfig.TopLeftY = 9;
    roiConfig.BotRightX = 9;
    roiConfig.BotRightY = 6;
    status += VL53L1_SetUserROI(dev, &roiConfig);

    uint32_t tbudget, inter;
    status += VL53L1_GetMeasurementTimingBudgetMicroSeconds(dev, &tbudget);
    status += VL53L1_GetInterMeasurementPeriodMilliSeconds(dev, &inter);
    DEBUG_PRINT("budget: %d, inter: %d\n", (int)tbudget, (int)inter);

    status += VL53L1_StartMeasurement(dev);
    return status == VL53L1_ERROR_NONE;
}

static void mrTask(void *param) {
    systemWaitStart();
    TickType_t lastWakeTime = xTaskGetTickCount();
    while (1) {
        vTaskDelayUntil(&lastWakeTime, M2T(30));
        uint16_t front = mrGetMeasurementAndRestart(&devFront);
        // DEBUG_PRINT("front %d\n", (int)front);
        rangeSet(rangeFront, front/1000.0f);
        rangeSet(rangeBack, mrGetMeasurementAndRestart(&devBack)/1000.0f);
        rangeSet(rangeUp, mrGetMeasurementAndRestart(&devUp)/1000.0f);
        rangeSet(rangeLeft, mrGetMeasurementAndRestart(&devLeft)/1000.0f);
        rangeSet(rangeRight, mrGetMeasurementAndRestart(&devRight)/1000.0f);
    }
}

static void mrInit() {
    if (isInit)
        return;

    isInit = true;
    pca95x4Init();
    pca95x4ConfigOutput(
        ~(MR_PIN_UP | MR_PIN_RIGHT | MR_PIN_LEFT | MR_PIN_FRONT | MR_PIN_BACK));
    pca95x4ClearOutput(MR_PIN_UP | MR_PIN_RIGHT | MR_PIN_LEFT | MR_PIN_FRONT |
                       MR_PIN_BACK);

    pca95x4SetOutput(MR_PIN_FRONT);
    if (setupSensor(&devFront, I2C1_DEV)) {
        DEBUG_PRINT("Init front sensor [OK]\n");
    } else {
        DEBUG_PRINT("Init front sensor [FAIL]\n");
        isInit = false;
    }

    pca95x4SetOutput(MR_PIN_BACK);
    if (setupSensor(&devBack, I2C1_DEV)) {
        DEBUG_PRINT("Init back sensor [OK]\n");
    } else {
        DEBUG_PRINT("Init back sensor [FAIL]\n");
        isInit = false;
    }

    pca95x4SetOutput(MR_PIN_UP);
    if (setupSensor(&devUp, I2C1_DEV)) {
        DEBUG_PRINT("Init up sensor [OK]\n");
    } else {
        DEBUG_PRINT("Init up sensor [FAIL]\n");
        isInit = false;
    }

    pca95x4SetOutput(MR_PIN_LEFT);
    if (setupSensor(&devLeft, I2C1_DEV)) {
        DEBUG_PRINT("Init left sensor [OK]\n");
    } else {
        DEBUG_PRINT("Init left sensor [FAIL]\n");
        isInit = false;
    }

    pca95x4SetOutput(MR_PIN_RIGHT);
    if (setupSensor(&devRight, I2C1_DEV)) {
        DEBUG_PRINT("Init right sensor [OK]\n");
    } else {
        DEBUG_PRINT("Init right sensor [FAIL]\n");
        isInit = false;
    }

    xTaskCreate(mrTask, MULTIRANGER_TASK_NAME, MULTIRANGER_TASK_STACKSIZE, NULL,
                MULTIRANGER_TASK_PRI, NULL);
}

static bool mrTest() {
    return isInit;
}

static const DeckDriver multiranger_deck = {
    .vid = 0xBC,
    .pid = 0x0C,
    .name = "bcMultiranger",

    .usedGpio = 0,  // FIXME: set the used pins

    .init = mrInit,
    .test = mrTest,
};

DECK_DRIVER(multiranger_deck);

PARAM_GROUP_START(deck)
PARAM_ADD(PARAM_UINT8 | PARAM_RONLY, bcMultiranger, &isInit)
PARAM_GROUP_STOP(deck)
```

hanzhi713, 2021-03-12:
By the way, both the current version and my modified version of the driver **do not work** on the latest master code. All sensor inits will `[FAIL]`. 

I'm able to make both work on tag 2021.01.

tobbeanton, 2021-03-18:
> By the way, both the current version and my modified version of the driver **do not work** on the latest master code. All sensor inits will `[FAIL]`. Is it related to the recent i2c changes?
> 
> I'm able to make both work on tag 2021.01.

We cleaned up the I2C driver a bit and it changed timing. The VL53 now needs a sleep after it is woken up until it is initialized. We plan to push a fix today.

tobbeanton, 2021-03-18:
The VL53-down is used for height stabilization which works better at a higher frequency but that means less accuracy, which is the trade-off.
```
  VL53L1_SetDistanceMode(&dev, VL53L1_DISTANCEMODE_MEDIUM);
  VL53L1_SetMeasurementTimingBudgetMicroSeconds(&dev, 25000);

```
It is possibly to adjust the multiranger sensors to you liking. 

I had problems with the VL53L1_ClearInterruptAndStartMeasurement as well. If you find out why it doesn't work let us know!

hanzhi713, 2021-03-19:
In my experiment, a timing budget of 25ms indeed brings more noise to the data, and sometimes the OS does not schedule enough ticks to the multirange driver, causing the actual ranging frequency to decrease. I found that 40~45ms timing budget gives accuracy comparable to 100ms (which is the default now) while having double the data. 

Also, good to know that the i2c issue is fixed.

hanzhi713, 2021-03-19:
I think I'm all good for now. If I have any new discovery regarding the issue of VL53L1_ClearInterruptAndStartMeasurement I'll let you know. Thanks for the explanation.

closed 2021-03-19 by hanzhi713