bitcraze/crazyflie-firmware #740: Flight Control

Sigmund-SJTUer, 2021-03-19:
I wanna control Crazyflie2.1 without flowdeck and multiranger. But MotionCommander lib is designed on basis of these two modules. How should I do?

knmcguire, 2021-03-19:
Hi! This is difficult to do autonomously since you need a positioning system for this. Please checkout our forum.bitcraze.io if anybody tried this before.

Best regards,

Kimberly

Sigmund-SJTUer, 2021-03-21:
We have a optical capture positioning system called VICON. I wanna use the feedback data of this system to control crazyflie2.1. Is there some examples to be referred?

whoenig, 2021-03-21:
The Crazyswarm has integrated VICON support: https://crazyswarm.readthedocs.io/. Another option is to use the python-lib and the VICON DataStream SDK.

closed 2021-03-22 by Sigmund-SJTUer