bitcraze/crazyflie-firmware #762: Crazyflie CRTP_Param Documentation is outdated

Steven-Sorokin, 2021-04-25:
[This wiki page](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/functional-areas/crtp/crtp_parameters.md) is outdated. There is no reset command `0`  that I have found in [the python library toc.py](https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crazyflie/toc.py), and the commands stand for different things in the python code compared to what is written in the wiki.

[image]

[image]

jonasdn, 2021-04-27:
Hi Steven-Sorokin!

Yes, the wiki is outdated, we are moving away from it and towards documentation on bitcraze.io and on GitHub.

What are you trying to do using the CRTP protocol? Maybe I can point you in the right direction.

Allt the best
Jonas

jonasdn, 2021-04-27:
Yes, that seems to be a bit out-of-date, thank you! Are you trying to write something using the CRTP protocol?

trakevital, 2021-06-12:
Yes! Sorry for the very late reply but I'm currently working on a C++ implementation of the crazyflie lib python. It's still in development but you can see it at https://github.com/M-R-VulcaN/crazyflie-link-cpp-expanded.

jonasdn, 2021-09-03:
What is the plan for this issue?

Should we:

a. Make an effort, add a story, and update all of the CRTP documentation in markdown
b. Remove the CRTP documentation markdown 
c. Add doxygen comment of the CRTP documentation
d. Move/add CRTP doxygen documentation to another repo? (website)?

whoenig, 2021-09-06:
This spans at least two repos (NRF51 and STM32 firmware), so perhaps d) is the appropriate action here?

knmcguire, 2021-09-06:
The idea that we had is that the STM repo is actually the one that acts upon the CRTP message and therefore it should be there, at least that the initial assumption that we went with when we all first started moving the wiki to the repo docs.

Since moving the CRTP doc to another place or the website is part of a bigger discussion, I kind of vote of just resolving this issue in the repository doc itself, so option A as  @jonasdn proposed

ataffanel, 2021-09-07:
I do not think CRTP spans on the nRF51 and STM32 repos, this is not intended at least: The nRF51 implements the link and provides a bidirectinal stream of CRTP packets to the STM32 (it acts as a bridge). STM32 is the one implementing the API, the CRTP packet content handling. So I still think having the CRTP docutmentation in the STM32 repos makes sense.

The nRF51 do implement a couple of packets, these packets are called bootloaderCmd and radioCmd. They are sent independently of safelink and so are kind of off-band of CRTP. I would say that they should be documented in the crazyflie2-nrf-firmware.

It does split the documentation but we can still link the two and I think that it still makes some sense: when communicating with the Crazyflie over USB or Serial port, all you have access to is the CRTP packets going to the STM32, not to the radio off-band packets, so they should not be on the same page of doc anyway.

As for the original topic of this ticket: I have experienced making a Rust Crazyflie lib recently and I can confirm that the CRTP doc needs a big update. Let's get to it!

zanecodes, 2025-01-28:
The CRTP parameter documentation is still outdated, #940 did not address this. Sending CRTP packets with the message IDs shown in [the documentation](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/crtp/crtp_parameters/#toc-access) results in a `Param API V1 not supported anymore!` debug response from the Crazyflie.

closed 2026-09-02 by enyanil