bitcraze/crazyflie-firmware #765: Build error with fresh repo clone

bangrock, 2021-04-28:
Hi,

I started with a fresh clone recently and I ran into this error while building. I am using the VM from the documentation.

  CLEAN_VERSION
  CC    position_controller_indi.o
.//src/modules/src/position_controller_indi.c: In function 'positionControllerINDI':
.//src/modules/src/position_controller_indi.c:180:2: error: missing braces around initializer [-Werror=missing-braces]
  float M_OB[3][3] = {0};
  ^
.//src/modules/src/position_controller_indi.c:180:2: error: (near initialization for 'M_OB[0]') [-Werror=missing-braces]
.//src/modules/src/position_controller_indi.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: all warnings being treated as errors
make[1]: *** [position_controller_indi.o] Error 1
make: *** [build] Error 2

I can fix the error by adding another set of braces around the zero but then I run into this error.

.//src/modules/src/lighthouse/lighthouse_position_est.c: In function 'estimatePositionSweepsLh1':
.//src/modules/src/lighthouse/lighthouse_position_est.c:288:27: error: assignment from incompatible pointer type [-Werror]
       sweepInfo.sensorPos = &sensorDeckPositions[sensor];
                           ^
.//src/modules/src/lighthouse/lighthouse_position_est.c: In function 'estimatePositionSweepsLh2':
.//src/modules/src/lighthouse/lighthouse_position_est.c:331:27: error: assignment from incompatible pointer type [-Werror]
       sweepInfo.sensorPos = &sensorDeckPositions[sensor];
                           ^
.//src/modules/src/lighthouse/lighthouse_position_est.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: all warnings being treated as errors
make[1]: *** [lighthouse_position_est.o] Error 1
rm version.c
make: *** [build] Error 2

I am not sure how to fix this

jonasdn, 2021-04-29:
Hi @bangrock!

Lets get to the bottom of this! I would like to try to reproduce this, so I need to know the _exact_ steps you took to get into this mess!

I need to ask you some annoying questions:

1. You downloaded the VM from bitcraze release github?
2. Which version?
3. What did you do after the VM booted?
4. What commands did you type in order to get a fresh clone of the firmware?
5. How did you build it? (Which commands)

My attempt to reproduce this so far:

1. Downloaded Bitcraze VM from here: https://github.com/bitcraze/bitcraze-vm/releases/
2. Version 2021.04: (https://storage.googleapis.com/bitcraze-share/vm/BitcrazeVM_2021.04.ova)
3. Ran "Update All Projects" from the desktop
  * Started a terminal and made sure the current source copy in the VM builds.
```
$ cd projects/crazyflie-firmware
$ make
CLEAN_VERSION
cc list.o
[...]
LD    cf2.elf
  COPY  cf2.hex
  COPY  cf2.bin
  DFUse cf2.dfu
Build for the CF2 platform!
Build 35:d0ca74f3692e (2021.03 +35) CLEAN
Version extracted from git
Crazyloader build!
Flash |  405868/1032192 (39%),  626324 free | text: 399724, data: 6144, ccmdata: 0
RAM   |   71640/131072  (55%),   59432 free | bss: 65496, data: 6144
CCM   |   58524/65536   (89%),    7012 free | ccmbss: 58524, ccmdata: 0
```

4. Do a fresh clone:
```
$ cd ~
$ git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git
[...]
```
5. Build firmware
```
$ cd crazyflie-firmware
$ make -j 12
CLEAN_VERSION
list.o
[...]
LD    cf2.elf
  COPY  cf2.hex
  COPY  cf2.bin
  DFUse cf2.dfu
Build for the CF2 platform!
Build 35:d0ca74f3692e (2021.03 +35) CLEAN
Version extracted from git
Crazyloader build!
Flash |  405868/1032192 (39%),  626324 free | text: 399724, data: 6144, ccmdata: 0
RAM   |   71640/131072  (55%),   59432 free | bss: 65496, data: 6144
CCM   |   58524/65536   (89%),    7012 free | ccmbss: 58524, ccmdata: 0
```

All this worked for me...

bangrock, 2021-04-29:
I was using VM version 2017.03

I changed it to the version you suggested and it works perfectly

Thank you for the help

closed 2021-04-29 by bangrock