bitcraze/crazyflie-firmware #771: "Disable" flow deck measurings in the Kalman estimator.

santiagorg2401, 2021-04-30:
Hi, I hope you're all well,
I am developing a project of obstacle detection, the sensors are the LPS System (8 nodes) and the flow deck, for developing it I need that the CF "ignores" the flow deck readings for its state estimator, however, I will be comparing the laser readings with the state estimator (sensors without flow deck) in order to see if there is an obstacle or not,.

I am developing this application on a ROS Node, on which I can access and compare the readings, what I do not know Is how to make the Kalman estimator to not take in count the flow deck readings, do you know how to diable the flow deck measurings in the state estimator?

Thank you in advance.

knmcguire, 2021-05-03:
Hi  @santiagorg2401, Seems that you have fixed this issue already. Could you let us know how you managed to solve the problem?

santiagorg2401, 2021-05-03:
Hello, of course, I modified the files: 

estimator.c
estimator_kalman.c
kalman_core.c
range.c
flowdeck_v1v2.c
zranger.c
zranger2.c

Then I flashed it to the Crazyflie,  I haven't tested it yet with the LPS System due from blockades in the area, however, it seems to worked.
The flow decks measurements were not exactly necessary to be changed, nevertheless I decided to do so to avoid problems with the surface.
///Modified files:
[modifiedFiles.zip](https://github.com/bitcraze/crazyflie-firmware/files/6415157/modifiedFiles.zip)

knmcguire, 2021-05-03:
Ah thanks for letting us know. It is still an interesting use case still, also a bit related to this [pull request for the lighthouse](https://github.com/bitcraze/crazyflie-firmware/pull/768) deck. Maybe more people want to have this so I'll keep this issue open for now.

I removed your fixed, since this is a feature that is not intergrated into the firmware yet ;) but perhaps we will in the future.

santiagorg2401, 2021-05-03:
I understand, thank you.

knmcguire, 2021-05-12:
Actually this already exists! The parameter motion.useFlowDisabled on 1 will prevent the flow to go to the kalman filter. I will close the issue.

closed 2021-05-12 by knmcguire

santiagorg2401, 2021-07-21:
Hi @knmcguire , I hope you're fine,
I saw the new firmware changes since I created this issue, however I really got lost trying to disable the flow and range readings into the EKF, could you please tell me how should I approach the problem?

Best regards.

knmcguire, 2021-07-22:
You can disable the flow measurements being send to the EKF by putting the parameter motion.useFlowDisabled on 1. This can be done from 

1- cfclient from the parameter tab ([Here instructions](https://www.bitcraze.io/documentation/repository/crazyflie-clients-python/master/userguides/userguide_client/#tabs) to reach that tab)
2- A python script with the cflib python library. [Here are the instructions](https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/user-guides/python_api/#parameters) of how to set a paramter. If that is still unclear, I advise you to take the [step by step guide for logs and parameters](https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/user-guides/sbs_connect_log_param/).

santiagorg2401, 2021-07-23:
Does it include the laser measurements as well?

ataffanel, 2021-07-26:
No, the parameter [motion.disable](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#motiondisable) will only disable the motion sensor input to the EKF. There is, as far as I know, no way to disable the z-ranger via parameters, it has to be done in the code.

For reference, the Z sensor is being pushed to the EKF in zranger2.c: https://github.com/bitcraze/crazyflie-firmware/blob/cbbbbdee7ad6fd4cf9d50dff722a7693bee181b5/src/deck/drivers/src/zranger2.c#L143