bitcraze/crazyflie-firmware #785: Flag ENABLE_UART1_DMA leads to compile error

bot-motion, 2021-05-22:
I'd like to use DMA on UART1 on the CF2.1. 

Simply switching the compiler flag ENABLE_UART1_DMA in uart1.c leads to a conflicting definition in the Bolt driver for the BMI388:

[image]

I didn't want to spend time on this, so I hacked my way around it. Is there a proper way to activate DMA for UART1?

bot-motion, 2021-05-29:
Thank you for your answer! Maybe I misinterpreted the comment in uart1.c, where it says/said sth along the lines
```
/** This uart is conflicting with SPI2 DMA used in sensors_bmi088_spi_bmp388.c
*  which is used in CF-Bolt. So for other products this can be enabled.
*/
#define ENABLE_UART1_DMA
```
Since I don't have a bolt, I thought that the SPI connection for BMI/BMP is not used at all? Is that correct? 

The problem seemed to me that the firmware compiles the drivers for the sensors in their hardware setup "CF" and "bolt" into the same bin file in `platform_cf2.c`. 

I did something very ugly: I duplicated the `DMA1_Stream3_IRQHandler` code, renamed one to `..._2` and attached that to the SPI. It would be nicer to introduce a compile flag... if I understand correctly, the tradeoff in this case would have been "can I detect the hardware platform I'm on when I'm running - then I need all the drivers for all variants compiled in" or "do I have to tell what hardware platform to compile for but then I can never run the same firmware on different hardware configs".

Anyway, thanks again - if you could just confirm that the SPI is unused for the BMI, I'd be very grateful, and you could close this!

tobbeanton, 2021-06-01:
Ahh, I missed that you are using the CF2.1. Then it should work, but as you write, you will have to remove the DMA1_Stream3_IRQHandler/BMI088_SPI_RX_DMA_IRQHandler function in sensors_bmi088_spi.c. This could be removed automatically with a define flag but that could also be dangerous if the user does not understand that.

closed 2021-08-30 by jonasdn