bitcraze/crazyflie-firmware #796: assert fail if log groupname and variable name are identical

wydmynd, 2021-06-17:
I do not have the assert information but very easy to replicate

krichardsson, 2021-06-17:
Thanks for the report! 

Could you please add an example config that causes the problem to make it easy to reproduce

tomso1, 2021-06-17:
I think you misunderstood me, if you define a log variable in your code , and the group name and variable name are identical, 
The CF will start correctly, but upon starting to log this variable to SD, there will be an assert fail. I will try to create an example.

krichardsson, 2021-06-17:
Ah, I see! An example is good to make it clear.
Thanks!

jonasdn, 2021-06-17:
Hi tomso1!

This is expected! The way we add logs and params via a magic macro (https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/interface/log.h#L168) that adds the parameter group and name at a special point in memory makes it hard for us to detect duplicates at compile-time.

We have a script at `tools/build/elf_sanity.py` that checks for duplicates when our GitHub CI runs.

If I add a duplicate:

```
diff --git a/src/deck/drivers/src/activeMarkerDeck.c b/src/deck/drivers/src/activeMarkerDeck.c
index 27ca638c..ff34565b 100644
--- a/src/deck/drivers/src/activeMarkerDeck.c
+++ b/src/deck/drivers/src/activeMarkerDeck.c
@@ -309,5 +309,6 @@ PARAM_GROUP_STOP(deck)
 
 LOG_GROUP_START(activeMarker)
 LOG_ADD(LOG_UINT8, btSns, &deckButtonSensorValue)
+LOG_ADD(LOG_UINT8, btSns, &deckButtonSensorValue)
 LOG_ADD(LOG_UINT8, i2cOk, &i2cOk)
 LOG_GROUP_STOP(activeMarker)
```

And run the script:
```
$ python3 tools/build/elf_sanity.py cf2.elf 
Duplicate parameter detected! (activeMarker.btSns)
```

It catches this. So in the official firmware this should not be an issue. But when adding your own one need to be careful.

Thanks
Jonas

closed 2021-06-17 by jonasdn