bitcraze/crazyflie-firmware #798: usd unable to log

easy98, 2021-06-20:
Hi,

My usd is not able to log any logging variables due to the assert problem here(https://github.com/bitcraze/crazyflie-firmware/blob/b448553725634aa83d25b356a7de12ba61647377/src/deck/drivers/src/usddeck.c#L504). I use the latest firmware.

> SUMMARY
========

PARAMETERS
 * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/CF21SingleMarker/bigQuad: False
 * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
 * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/defaultSingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/defaultSingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/defaultSingleMarker/bigQuad: False
 * /crazyflieTypes/defaultSingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/defaultSingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
[... 199 lines cut ...]
[ INFO] [1624232704.340154824]: [cf4] uSD: wait for sensors
[ WARN] [1624232704.372271294]: [cf4] Link Quality low (0.630000)
[ INFO] [1624232705.337373600]: [cf4] IMU: BMI088 Gyro connection [OK].
[ INFO] [1624232705.391946324]: [cf4] IMU: BMI088 Accel connection [OK]
[ INFO] [1624232705.405754023]: [cf4] IMU: BMP388 I2C connection [OK]
[ INFO] [1624232705.426961765]: [cf4] ESTIMATOR: Using Complementary (1) estimator
[ INFO] [1624232705.429507587]: [cf4] CONTROLLER: Using PID (1) controller
[ INFO] [1624232705.431969630]: [cf4] MTR-DRV: Using brushed motor driver
[ INFO] [1624232705.434565329]: [cf4] SYS: About to run tests in system.c.
[ WARN] [1624232705.627962767]: [cf4] Link Quality low (0.030000)
[ INFO] [1624232705.635151848]: [cf4] SYS: ----------------------------
[ INFO] [1624232705.660259452]: [cf4] SYS: Crazyflie 2.1 is up and running!
[ INFO] [1624232705.663078305]: [cf4] SYS: Build 215:a7e0aae54b23 (2021.03 +215) CLEAN
[ INFO] [1624232705.684592779]: [cf4] ash!
[ INFO] [1624232705.686133906]: [cf4] CFGBLK: v1, verification [OK]
[ INFO] [1624232705.709374056]: [cf4] DECK_CORE: 1 deck(s) found
[ INFO] [1624232705.712371165]: [cf4] DECK_CORE: Calling INIT on driver bcUSD for deck 0
[ INFO] [1624232705.721803673]: [cf4] uSD: mount SD-Card [OK].
[ INFO] [1624232705.724346298]: [cf4] IMU: BMI088: Using I2C interface.
[ INFO] [1624232705.725845507]: [cf4] uSD: wait for sensors
[ WARN] [1624232705.729431289]: [cf4] Link Quality low (0.330000)
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyswarm_server-2] process has died [pid 29733, exit code -6, cmd /home/coven-lab/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/coven-lab/.ros/log/887091b2-d221-11eb-9720-606dc7c60f33/crazyswarm_server-2.log].
log file: /home/coven-lab/.ros/log/887091b2-d221-11eb-9720-606dc7c60f33/crazyswarm_server-2*.log

krichardsson, 2021-06-21:
Since you are using the 2021.03 release, the assert is actually here https://github.com/bitcraze/crazyflie-firmware/blob/2021.03/src/deck/drivers/src/usddeck.c#L504

krichardsson, 2021-06-21:
It seems to be related to the CORE log implementation we did lately. Also see #791

jonasdn, 2021-06-21:
Hi!

Could you perhaps share your log configuration? What variables are you trying to log?

easy98, 2021-06-21:
Hi,

It's my configuration file:

> 1     # version
> 512   # buffer size in bytes
> log   # file name
> 1     # enable on startup (0/1)
> on:fixedFrequency
> 250     # frequency
> 1     # mode (0: disabled, 1: synchronous stabilizer, 2: asynchronous)
> 
> acc.x
> acc.y
> acc.z
> gyro.x
> gyro.y
> gyro.z
> baro.asl
> baro.temp
> baro.pressure
> stabilizer.roll
> stabilizer.pitch
> stabilizer.yaw
> stabilizer.thrust
> ctrltarget.roll
> ctrltarget.pitch
> ctrltarget.yaw

jonasdn, 2021-06-22:
I think the latest PR (#800) should fix this, are you able to test it @easy98 ?

easy98, 2021-06-22:
Thank you for your reply!

I change the PR. Now it can create a log file but it's empty and unable to log anything.

easy98, 2021-06-25:
@jonasdn Hi, I update the PR and do some tests. Now the config file can be read  and a log file can be created in the sd-card. But it's totally empty. I am so confused about it and don't know what might be the problem. Do you have any suggestions? Thanks!

closed 2021-07-01 by jonasdn

sstroobants, 2023-03-12:
Hi, has this ever been resolved? I think I have the same issues. As soon as I set the usd.logging parameter in the cfclient to 0, it writes the data to file.

krichardsson, 2023-03-13:
@sstroobants Please open a new issue if you think you have found a bug or start a discussion (in http://discussions.bitcraze.io/) if you need support. Also refer to this issue if you think there is a relation.

PR #799 (2021-06-22): usddeck: Add eventriggerType_fp16
Fixes: #798
PR #800 (2021-07-01): log: Use TYPE_MASK to avoid problems with LOG_CORE
Fixes: #798
commit a096b497 (2021-07-01): log: Use TYPE_MASK to avoid problems with LOG_CORE (#800)

* log: Use TYPE_MASK to avoid problems with LOG_CORE

Fixes: #798

* log: Use logGetType() more

Instead of open coding it.