bitcraze/crazyflie-firmware #801: Proper usage of crtpSendPacket

jdavidberger, 2021-06-23:
I'm trying to use `crtpSendPacket` in a custom module and can't figure out the proper way to do it safely.

From [here](https://github.com/bitcraze/crazyflie-firmware/blob/ffc66a77414ca5355f579a0ab5facdaeea926e4a/src/modules/src/crtp.c#L215) it takes the given pointer and queues it to a different task; but there seems to be no way to know when that task is done with it. This is an issue since I want to send many packets at a relatively high throughput -- I won't know when it is safe to reuse the buffer. I was expecting either some form of callback or even just a 'sent' flag on the packet and found neither. 

For what it is worth, I haven't run into this issue in practice; and possibly there is something intrinsic about it that makes the chances of a race condition here go from pretty small to nil? 

If this is an actual issue I can make a PR that adds a flag to the packet to act as an indication that its busy or something.

ataffanel, 2021-06-23:
The FreeRTOS queues are pretty primitive in the sense that they `memcopy` the object. This is not the most efficient but it is by far the safest and easiest. So in the case of `crtpSendPacket`, if there was space in the queue, when the function returns your packet has been deep-copied into the FreeRTOS queue and you can safely re-use it for something else.

If you plan to send a lot of data, one interesting things to look at is that we have `crtpSendPacket` and `crtpSendPacketBlock`. The former will never block but might return an error if the send queue was full. The later will block your task until the packet has been copied in the queue. Depending of your implementation one might be better than the other. In the Crazyflie we normally use the blocking version for 'safe' packets that must be sent and the non-blocking version for things that are best-effort (like the log packets).

jdavidberger, 2021-06-23:
Oops, my apologies -- I had it in my head that freertos had zero copy by default behavior on it's queues. Not really sure why. 

I did see that and was going to use the non blocking version, and then just track if it was getting majorly squelched via the logging framework. 

Thank you!

closed 2021-06-23 by jdavidberger