bitcraze/crazyflie-firmware #806: Problem with combo Lighthouse deck and Flowdeck

knmcguire, 2021-06-24:
The combination of the flowdeck and lighthouse deck on current master has a problem! 
'
With both attached to the crazyflie, the (newly flashed) lighthouse deck will show a blue LED. Also, without a base-station, the flowdeck's measurement seem not to be enough to converge the EKF estimated.

Using the decks separately does not cause this error and show normal behavior.

knmcguire, 2021-06-24:
UPDATE: I see the same thing with the LPS deck. No weird LED but a much slower update rate of the flowdeck's measurements

keffklau, 2021-06-28:
It might be another issue, but due to the fact that the zranger/ tof measurement is received by i2c: We also have issues using the MR-Deck in combination with lighthouse or LPS

knmcguire, 2021-07-09:
Seems like more people are having problems with it... https://forum.bitcraze.io/viewtopic.php?f=16&t=4954

closed 2021-08-11 by knmcguire