bitcraze/crazyflie-firmware #815: Assert if measurements of the IMU or deck stops

knmcguire, 2021-07-12:
PR #810 will perhaps now make it more difficult to spot failures where the IMU measurements stall or stop. We need to make an assert that not only checks the incoming of measurements of the IMU but also from expansion decks as well. Perhaps if it does not come in for 1 second?

This case, problems like #806 will also be easier to spot

jonasdn, 2021-09-03:
To solve this we should implement what?

  - A timer to measure delays im IMU measurements
  - An assert if delay is above a threshold

Is this an ok plan of attack?

knmcguire, 2021-12-22:
I've tried to look into this, but before we continue, maybe we should talk about where to place this assert first.

1. In the sensor driver itself, which then needs to be duplicated.
2. In the Enqueue part of the estimator? Making use of a timestamp that we send with it
3. Setup an state Rate counter for the IMU measurements that come in estimator, and assert when this rate is underneath a certain value.
4. Make a supervisor module that keeps all the timestamps in check, perhaps also for other measurements (but with less drastic results if it aint essential.

3 would be easy to setup but 2 can work too. perhaps since @whoenig made some changes to the estimator structure, maybe he has some opinions on this as well.

whoenig, 2021-12-23:
I actually think option 1 makes most sense - this seems to be a driver feature to me. Estimators don't really care about the rate of measurements (the covariance just becomes very large if you don't get a measurement for a while).

Option 4 would also be possible, if we would have a general health task that looks at statistics like that and warns the user if it detects problems.

I personally don't think code duplication is always a bad thing. From an engineering point of view, it is easy to handle if the files are similar using a good diff tool. From en embedded point of view, this can increase the code size, but I don't think that a check like you are describing would result in significant bloat.

knmcguire, 2021-12-23:
Thanks for answering! 

For 1 there only will be two duplications for the two IMU's (we can skip the bosch deck), so it won't be too bad indeed. And the flowdeck does something similar with timing as well. 

For 4 it might be interesting to consider in the grand scheme of things, as it is also handy to know for positioning measurement in general as well. Although I wouldn't want to say to shut off the props immediately, as I would advise for no IMU input, but perhaps it might be possible for the user to choose how the crazyflie reacts on these circumstances (prop off for no IMU measurement, or just a land or prop slowdown for loss of positioning). 

Anyway, the latter is perhaps food for discussion in another issue 😄 I think for this one I'll implement something simple for now with a console warning and an assert.

knmcguire, 2021-12-27:
made a draft PR for one driver in #909