bitcraze/crazyflie-firmware #818: Possible crash when Lighthouse BS version mismatch Crazyflie configuration

ataffanel, 2021-07-19:
There has been report of crash when lighthouse V1 are active but a Crazyflie with a lighthouse deck is currently setup for V2 system: https://forum.bitcraze.io/viewtopic.php?p=22730#p22730.

This is likely a bug in the pulse-processing part of the Crazyflie misinterpreting V1 signals in a way that generates a math error (division by 0?).

knmcguire, 2021-08-11:
I've tried to recreate this in the lab, constantly switching between LHv1 and v2... but no cigar. Perhaps this new pull request is preventing the assert #810 ?

closed 2021-09-01 by jonasdn

AkashJohnSubash, 2023-02-04:
Observed issue on CF2.1 with the 2022.12 release firmware, and 2x lighthouse 2 base stations
The log was identical to the one posted in the above linked old forum.

Results:
-Rebooted with both base stations off, reflashed firmware.  SYS: cfAssertNormalStart [FAIL]  

-Rebooted with base station 1 off . SYS: cfAssertNormalStart [FAIL]  

-Rebooted with base station 2 off . No error, but take-off not stable  

- Strangely my colleagues with their CF2.1 using the same firmware version, lighthouse deck version, base stations did not have this issue.

-Possibly due to a slight movement of base station 2, after my previous calibration
Following instructions by "oryx" in  the linked forum regarding the line of sight fixed the issue