bitcraze/crazyflie-firmware #819: Crazyflie Bolt: pm.state always reports that the drone is on charger

ntamas, 2021-07-28:
It looks like the current firmware erroneously reports that the drone is on charger (even if it isn't) when using it on a Crazyflie Bolt. The `pm.state` variable is equal to 2 when no USB cable is connected and the drone is sitting idle on the desk with a charged battery. On a stock Crazyflie, `pm.state` is equal to 0 in this case. As a consequence, `system.canFly` is zero and the Crazyflie Python client disables the command-based flight mode buttons.

ntamas, 2021-07-30:
Further investigation shows that this happens only with the 1-cell Li-ion prototype (as outlined in [this blog post](https://www.bitcraze.io/2020/10/testing-crazyflie-bolt-and-1-cell-li-ion/) ) but not with a similar build that has a two-cell Li-Po battery.

knmcguire, 2021-08-04:
hi! this is probably a topic for @tobbeanton but he is still on a holiday for another week. I'll assign him so that he can investigate once he is back

tobbeanton, 2021-09-07:
Issue confirmed! The problem most likely lies in the nRF part of the power management code. I will investigate further.

tobbeanton, 2021-09-07:
[nRF issue ](https://github.com/bitcraze/crazyflie2-nrf-firmware/issues/61) solved by setting flags to 0 if charger is not present.

closed 2021-09-07 by tobbeanton