bitcraze/crazyflie-firmware #821: Log and Param types not matching

trakevital, 2021-08-05:
I am currently working on a project with crazyflies and am using the crtp protocol. I encountered something today which I have just realized: The Log and Param type `#define`s do not match! 
[image]
[image]
 I would suggest you to change it in order to prevent confusion in the future. In my opinion the best implementation was in the param due to the user-freindly defines.

trakevital, 2021-08-05:
I understand this will be very difficult to change because many infrastructures were implemented using this method and so it probably won't be a good idea to change the implementation now...
Basically it's too late

closed 2021-08-05 by trakevital

ataffanel, 2021-08-05:
Thanks for the report but it is indeed way too late for this change. I can understand the frustration, overall the CRTP protocol needs to be documented much better.

This was actually done a bit on purpose: log and param are using the same TOC protocol. The type in the toc is specific to the subsystem that implements the TOC: it was envision that TOC could be used for more subsystems. So by having two different types, it forces implementation to treat the [type as a generic](https://github.com/ataffanel/crazyflie-lib-rs/blob/b701a4892803007a4c36576afd3595e687e26aa3/src/lib.rs#L156).