bitcraze/crazyflie-firmware #832: Mellinger controller: Attitude control unstable beyond about 30 degrees roll/pitch

whoenig, 2021-08-17:
the SE(3) controller from Mellinger/Li should be capable of trajectory tracking with around 60 degrees roll/pitch angle without problems. Our implementation only can get to around 30 degrees. This has been observed by @whoenig and @jpreiss in a trajectory tracking scenario and by @pPatrickK in https://github.com/USC-ACTLab/crazyswarm/issues/71 in a position setpoint step scenario. This might be due to some implementation bug or because of the low thrust-to-weight ratio of Crazyflies.

Transferred from https://github.com/USC-ACTLab/crazyswarm/issues/74.

jonasdn, 2021-09-03:
What would be a good way forward for this issue?

How do we determine if this is a bug or if we have reached a limitation of thrust-to-weight ratio?

Any ideas?

whoenig, 2021-09-06:
A software verification might be possible with the Python bindings (https://github.com/bitcraze/crazyflie-firmware/issues/602) and simulation of a model with a higher t2w ratio. A hardware verification would require upgraded motors.

whoenig, 2021-09-24:
One problem I just found is the omega error: this should be both in body coordinates (equation right above (8) in the paper; Lee uses R^T R_d, see (8) in his paper). I am not sure if this can cause these kind of errors, but it can certainly explain odd behavior for yaw !=0 cases.

jpreiss, 2021-09-27:
Did anyone ever do a comparison between the INDI controller and Mellinger for aggressive attitudes? If another controller performs significantly better, that would be evidence that this is a software bug.

knmcguire, 2021-09-27:
I've compared the PID and INDI controller, as the INDI controller should be able to withstand sudden distrurbances and windgusts, but not against Mellinger... 

Is there a trajectory that forces the 30 deg roll/pitch that we can compare the both with?

whoenig, 2021-09-27:
Executing https://raw.githubusercontent.com/USC-ACTLab/crazyswarm/master/ros_ws/src/crazyswarm/scripts/figure8.csv at 0.6 timescale results in 34 deg max roll and 23 deg max pitch. I remember that this results usually in crashes with the mellinger controller.

knmcguire, 2022-03-07:
This was very long over due.... but I finally got around to test this. I tested the trajectory on PID, Mellinger and INDI. Although the Crazyflie didn't crash, it did behave more aggressive and wobbly during the maneuver. With bad motors/propellers it probably crashed for sure. 

So I'm not sure what is causing it for Mellinger, but perhaps it would be good in general to implement default mellinger gains are a little bit downtuned. Anybody that has those values anywhere?

Videos (might need to downloads to see it)

**PID**

https://user-images.githubusercontent.com/11061260/157041677-6b549ae5-aa05-49e1-bb50-1799c26bb7e0.mp4

**Melinger**

https://user-images.githubusercontent.com/11061260/157041769-66b673c6-c770-4a51-afcc-c543b96bc65e.mp4

**INDI**

https://user-images.githubusercontent.com/11061260/157041788-7fb11563-a153-436c-b695-cedb9105b0ab.mp4