bitcraze/crazyflie-firmware #834: Crazyflie reboots when enabling `robustTdoa` parameter if firmware in debug mode

mendesk, 2021-08-19:
I'm having problems when trying to set `robustTdoa`mode while the firmware has DEBUG enabled. When setting that parameter to `1`, the drone will immediately reboot and the following lines are shown at the bottom of the console log:
```
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OKIMU: BMI088 gyro self-test [OK]
SYS: The system resumed after SYS: The system resumed after a failed assert [WARNING]
SYS: Assert failed at .//src/hSYS: Assert failed at .//src/hal/src/radiolink.c:164
SYS: cfAssertNormalStart [FAILSYS: cfAssertNormalStart [FAIL]
```
Setting `robustTwr` mode while in DEBUG is not a problem nor is setting 'robustTdoa` while not in DEBUG mode. Hardcoding `robustTdoa = false` in `estimator_kalman.c` instead of setting it through the parameter framework yields the same result. 

To reproduce from master branch or 2021.06 tag:
- Enable DEBUG mode (`echo DEBUG=1 > tools/make/config.mk`)
- Compile & flash
- Connect to crazyflie with python client and set parameter `robustTdoa`to 1

mendesk, 2021-08-20:
This was all done with the LPS anchors runnning in TDoA2 mode, when I switch them to TWR, I get these reboots when setting the `robustTwr`parameter instead (while in DEBUG, fine without)

jonasdn, 2021-08-31:
I can reproduce this, will look at it! Do you have any additional information for me?

mendesk, 2021-08-31:
I'm afraid not much. That the issue is also there when it's hardcoded makes it likely that it's in the firmware, had a look at the robust estimation code but couldn't immediately find something that would explain it only happening in DEBUG mode.

Due to time constraints I switched off DEBUG mode and continued with `robustTdoa` enabled. In the end I disabled that again though as I had the impression that while the estimations were quite stable, they seemed to be consistently off in the Y plane by about 20cms compared to the true location. (low accuracy, high precision) But I didn't have time to control for that nor do I have data to back it up, so take that with a grain of salt.

jonasdn, 2021-09-01:
A theory we have is that the robust versions of TWR/TdoA takes to much CPU, starving out the communication tasks and causing the assert / reboot.

If you have the time, could you try enabling DEBUG, but changing the Makefile so we still use `-Os?  instead of `-O0` ? To see if the quad behaves better?

knmcguire, 2022-03-07:
With the latest version of the firmware, I do not see this happening anymore. So I'll close it for now... let's see if this will not reappear again.

closed 2022-03-07 by knmcguire