bitcraze/crazyflie-firmware #854: A possibility to override and control the Crazyflie 2.X onboard LEDs from the ground

tobbeanton, 2021-09-23:
Sometimes it is wanted to control the onboard LEDs from e.g. a script rather then letting them show the build in functionality.
The Crazyflie 2.X has 6 controllable LEDs. 5 of them controlled by the STM32F405 and 1 controlled by the nRF51. As not all LEDs are controlled by the same MCU this complicates things a little.

The first step could be a stealth mode so they could all be off, but if they could be controlled individually this would give even bigger value.

tobbeanton, 2021-09-23:
This is a first shot at controlling the STM32F405 LEDs through parameters. It needs more work but basically works and could already now be tried out using the patch below.

[cf2_ext_led_control_base308ceb32.zip](https://github.com/bitcraze/crazyflie-firmware/files/7216181/cf2_ext_led_control_base308ceb32.zip)

To override LED control set the led.extCtrl parameter to 1. Setting it to 0 will restore normal functionality.
To control the LEDs set the led.bitpattern parameter. Each bit controls the LED to be on or off and the bit order is according to the led_t enum.
`typedef enum {LED_BLUE_L = 0, LED_GREEN_L, LED_RED_L, LED_GREEN_R, LED_RED_R} led_t;`
Thus setting led.pitpattern to 0x01 would turn on left blue (M3) LED and all others off.

jonasdn, 2021-09-23:
Coud you create a pull request from the patch in the zip?

jonasdn, 2021-11-15:
This is possible now on master

closed 2021-11-15 by jonasdn