bitcraze/crazyflie-firmware #884: About pid control rule

LESLIEWANGer, 2021-11-14:
Hi!
Recently I read the pid controller in CF's firmware(controller.c\pid.c and so on).I found a minus sign in line 114 in controller_pid.c and it confused me.As mentioned in the guidance,this plane follows the right-hand system so the Z-axis points up.Does it mean the "yaw rate data" from the imu points down as the same direction as gravity or any reason else?
Thank you!

[image]
[](url)

knmcguire, 2021-11-15:
Hi! 

Indeed according to the right hand convention the minus looks off. From historical reasons, the state and the following PID controller evolved that way, as probably it is easier for non-aerospace people to recognize a negative pitch as going down and a positive pitch as going up.  At least that's how I interpreted it.

The current coordinate system that depicts the state of the crazyflie can be found [here](https://www.bitcraze.io/documentation/system/platform/cf2-coordinate-system/), where you see that indeed the rotation around the y access is not according to the righthand notation and at one point, this probably will need to change. 

The problem is, is that changing these coordinates will have quite a lot of implications on a lot of systems within the firmware at this point as it needs to be changed simultaneously in all state estimators, controllers, documentation and perhaps even more.

LESLIEWANGer, 2021-11-16:
Thank You!
I understand the rule about pitch.
Actually I was going to ask some questions about the "yaw" direction. That's to say, what‘s the meaning of the minus before  "control->yaw"?
[image]
Thank you again!

knmcguire, 2021-11-17:
Ahh oke yeah I missed that line. It is indeed a bit odd so we will check it out... but it worked so far! We get back to you once we got a good look at it.

knmcguire, 2021-11-19:
To be completely honest, I can not find the reason why this yaw rate command is negative :confused: not in the cflib, or the gyroscope measurements. 

@matejkarasek you looked at the yaw controller quite closely recently. Any thoughts?

matejkarasek, 2021-11-22:
Yeah, this has something to do with the sign conventions...
The `control->yaw` is later used by the power distribution (power_distribution_stock.c), where the signals going to individual motors are generated. 
Just by looking at the power distribution formulas, it seems the sign convention for yaw of the power distribution is opposite to the CF2.X frame used elsewhere (ENU with opposite pitch):
https://www.bitcraze.io/images/documentation/overview/coordinate_system.jpg
E.g. motor 1 has a positive sign next to control->yaw. The M1 prop spins CCW when viewed from the top, so the reaction moment (which turns the crazyflie) will be CW. But positive yaw direction should be CCW...

Note the same sign change is done in Mellinger 
https://github.com/bitcraze/crazyflie-firmware/blob/e0bd5159ee012f6665816713ccc133b717ff2927/src/modules/src/controller_mellinger.c#L305
And also in INDI there are multiple places where the yaw sign is changed, e.g. here:
https://github.com/bitcraze/crazyflie-firmware/blob/e0bd5159ee012f6665816713ccc133b717ff2927/src/modules/src/controller_indi.c#L165

So, the current code is obviously correct but confusing...
We could:
1) make changes to the power distribution, such that the sign convention is consistent with the rest (ENU with opposite pitch)
2) Use a proper sign convention all across the code. ENU (with correct pitch) would likely be easier, NED would be what aerospace is using...

knmcguire, 2021-11-22:
Thanks for your elaborate answer @matejkarasek  Ah yeah I didn't think to look in the power distribution that closely but yes ofcourse you are right! Ofcourse the code is correct or else it would have not been flying... but it's confusing indeed. I do believe that we have to discuss internally what the right approach here.

Personally I like the ENU with reversed pitch but that is since I come originally from ground based robotics where the up z-axis makes sense. And then the reversed pitch makes more sense to me for control, but that might be quite personal (in videogames with any type of airplane I reverse the pitch always for me). Also we need to make a decision if we want to distinguish the crazyflie as a robotic platform that happens to fly, or a fullblown quadcopter type that is never going to touch the ground in any form. 

Anyway, this item has been marked by @jonasdn for a triage, so let's keep it open for that. I believe that we have provided enough info to make a decision about the next step. In the mean time, I hope that @LESLIEWANGer's questions have been resolved now?

jonasdn, 2021-12-07:
So, what we have said from Bitcraze is that is not something that we are likely to work actively on. We are aware of these issues, and that the current code is a result of _making stuff work_. There are room for _plenty_ of improvement to make the calculations, formulas and general flow of control more canonical.

We would be happy to look at contributions for people who want to make the code easier to follow!

knmcguire, 2022-03-07:
I'll close this as this issue is related to the discussion issue we had before: #396. There is a bunch of stuff that needs to be done.

closed 2022-03-07 by knmcguire