bitcraze/crazyflie-firmware #893: Managed to increase usddeck download speed by a factor of 10

trakevital, 2021-12-02:
I'm working on a project which uses the crazyflie usddeck to store log files and send those log files to the pc. A college of mine changed your code slightly which caused it to read the uSD card log by much more efficiently - by a factor of 10. Mind you this was back in January of 2021 so the usddeck code looks differently after these 10 months:

We added this line to the top of the code:
```c
bool logFileOpen = false;
```
And Changed the usddeckRead like so: 
```c
// Read "length" number of bytes at "offset" into "buffer" of current file
// Only works if logging is stopped
bool usddeckRead(uint32_t offset, uint8_t* buffer, uint16_t length)
{
  bool result = false;
  if (initSuccess && xSemaphoreTake(logFileMutex, 0) == pdTRUE) {
    if(!logFileOpen)// added by us
    {
      if (f_open(&logFile, usdLogConfig.filename, FA_READ) == FR_OK) 
        logFileOpen = true;// added by us
    }

    if (logFileOpen) // added by us
    {
      if (f_lseek(&logFile, offset) == FR_OK) 
      {
        UINT bytesRead;
        FRESULT r = f_read(&logFile, buffer, length, &bytesRead);
        
        if (r == FR_OK) 
        {
          result = true;
          if(f_tell(&logFile) >= f_size(&logFile))
          {
            f_close(&logFile);
            logFileOpen = false; // added by us
          }
        }
      } 
    }
    xSemaphoreGive(logFileMutex);
  }
  return result;
}
```

Your code:
```c

// Read "length" number of bytes at "offset" into "buffer" of current file
// Only works if logging is stopped
bool usddeckRead(uint32_t offset, uint8_t* buffer, uint16_t length)
{
  bool result = false;
  if (initSuccess && xSemaphoreTake(logFileMutex, 0) == pdTRUE) {
    if (f_open(&logFile, usdLogConfig.filename, FA_READ) == FR_OK) {
      if (f_lseek(&logFile, offset) == FR_OK) {
        UINT bytesRead;
        FRESULT r = f_read(&logFile, buffer, length, &bytesRead);
        f_close(&logFile);
        if (r == FR_OK && bytesRead == length) {
          result = true;
        }
      } else {
        f_close(&logFile);
      }
    }
    xSemaphoreGive(logFileMutex);
  }
  return result;
}
```

I believe the reason why we received such improved results was that instead of each time taking the semaphore and trying to reopen the file to see if it is being used, we just  added a flag which will act as a quick check if the file is busy right now and saves precious cpu time.

krichardsson, 2021-12-07:
Hi and thanks for the input!

The SD card deck functionality has mainly been designed for moving the SD-card to a computer when analyzing the data, not reading it back via radio (or USB). I think the read functionality mainly was intended to enable to get small data quantities for verification that it works.

In your use case, do you read data back via the radio?

trakevital, 2021-12-08:
> In your use case, do you read data back via the radio?

Yes more or less, we send all the chunks of data via the usb.

krichardsson, 2021-12-09:
OK, thanks for the info!

knmcguire, 2022-03-07:
@trakevital we invite you to do a contribution of this in the firmware if you have time for it. It's a nice functionality but we will not have the time ourselves to implement it.

trakevital, 2022-05-12:
> @trakevital we invite you to do a contribution of this in the firmware if you have time for it. It's a nice functionality but we will not have the time ourselves to implement it.

I'm afraid I have stopped working in the crazyflie project. I will send this pull request to an ex-college whose still working with crazyflies