bitcraze/crazyflie-firmware #897: IMU Position Offset Compensation #286

Moj-Dev, 2021-12-06:
How can I change the offset position of IMU sensors in the firmware if the IMU position is not exactly in the center of gravity?

krichardsson, 2021-12-09:
This type of functionality is unfortunately currently not implemented in the platform. On the Crazyflie 2.1 we have assumed it would not make any significant difference, but we have not really made any measurements though. 
Do you have a use case where you think it would be important? Could you tell us a bit about it?

knmcguire, 2022-03-07:
This is a bit related to this discussion: #937 and adding of new platforms #946. 

We first need to implement the IMU to have an offset (orientation has already been done with #829). But the offset will also require us to change things in the state estimate filters as well.. So it might be a bigger issue to fix though....

Let us discuss this internally.

knmcguire, 2022-09-06:
This is part of the bigger discussion about potentially restructuring the IMU handling.