bitcraze/crazyflie-firmware #900: eps

rtkartista, 2021-12-13:


rtkartista, 2021-12-13:
I just commented out 
1.  PROJ_OBJ += esp_deck_flasher.o
2. PROJ_OBJ += aideck.o

these lines in the makefile and the make was successful

knmcguire, 2021-12-16:
Hi! Happy you found a solution.

I just reopened as this looks like an issue that we need to address for this repository as we can't uncomment these lines by standard. So just for our investigation

knmcguire, 2021-12-23:
So we haven't seen this ourselves with the latest master of the code. Perhaps you just pulled a version that had this build error just at the right time. Also double check your gcc compiler version and if your submodules are all properly updated.

I'll close this issue now but please let us know if you still have this problem.

Moj-Dev, 2022-01-04:
> I just commented out
> 
> 1. PROJ_OBJ += esp_deck_flasher.o
> 2. PROJ_OBJ += aideck.o
> 
> these lines in the makefile and the make was successful

I have the same problem. Did you succeed to fix the issue? Does your crazyflie work well? In my case, the kalman filter resets!

knmcguire, 2022-01-05:
Aha! I saw in a previous issue (#910) that you @Moj-Dev mentioned that you used Ubuntu 18.04 with arm-gcc version 7.3. I've used an VM with those some characteristics and indeed, it fails for me as well.:
```c
.//src/modules/src/esp_deck_flasher.c: In function 'espDeckFlasherWrite':
.//src/modules/src/esp_deck_flasher.c:136:21: error: 'lastPacketFull' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 if ((lastPacket && lastPacketFull))
 ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~ 
.//src/modules/src/esp_deck_flasher.c: At top level:
cc1: error: unrecognized command line option '-Wno-stringop-overread' [-Werror]
cc1: error: unrecognized command line option '-Wno-address-of-packed-member' [-Werror]   
```

My version of arm-none-eabi-gcc 
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
```

Good catch! we need to start fixing this as it would be good to still have ubuntu 18.04 as compatible

knmcguire, 2022-01-05:
Just as reference, for Ubuntu 20.04 and arm gcc version 9.2.1 the firmware builds fine:
```
$arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] 
```
So perhaps those older compilers do not recognize these flags:` -Wno-stringop-overread` and `-Wno-address-of-packed-member`. I don't see any protection for this in the crazyflie firmware's makefile..

knmcguire, 2022-01-05:
These flags were added 3 months ago with this PR: #877

closed 2022-01-27 by knmcguire