bitcraze/crazyflie-firmware #908: Doc: update PWM to thrust page with thrust stand measurements

knmcguire, 2021-12-22:
The PWM to thrust investigation page is now part of the[ documentation in this repository](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/pwm-to-thrust/), but this are pretty old measurements and must be at one point updated with the new thruststand measurements [from this blogpost](https://www.bitcraze.io/2021/08/building-a-crazyflie-thrust-stand/)

whoenig, 2021-12-22:
And/or Table 1 of https://whoenig.github.io/publications/2021_T-RO_Shi.pdf.

knmcguire, 2022-11-09:
I'm a bit nervous of adding plots of published papers. Also the formula needs some context as well of table 1. What I did now, that on top I've added a list of publications that did system ID with the crazyflie, where I added the pdf.

closed 2022-11-09 by knmcguire