bitcraze/crazyflie-firmware #910: ESTKALMAN: State out of bounds, resetting

Moj-Dev, 2021-12-29:
I'm getting "ESTKALMAN: State out of bounds, resetting" error after updating the bolt's firmware to the latest version. I appreciate it if you could help me out to find the problem. 

Thanks,

ataffanel, 2021-12-29:
This means that the kalman filter is enabled but no position information is fed to it. The Kalman filter is normally not enabled by default unless one deck is requesting it, do you have a positioning deck installed on the Bolt?

If so, feeding position information to the deck (the exact mean depends of the deck) should remove this message.

Moj-Dev, 2021-12-29:
No, there are no positioning decks!
Should I modify MakeFile in the firmware?

knmcguire, 2022-01-03:
Hi!

It's strange that it would default to the kalman filter if you don't have any positioning decks... I just checked this with the crazyflie with the latest firmware and that doesn't have the problem. 

What you could do is to force the estimator to be the complementary filter during compile. You can find the instructions here: https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/sensor-to-control/configure_estimator_controller/

Moj-Dev, 2022-01-04:
I guess the problem is related to gcc compiler version. I'm using version 7.5.0 in Ubuntu 18.04. Did you compile the firmware with this setting? 
In addition, since I want to feed the position data as external data to the crazyflie, I can't use the complementary filter.

knmcguire, 2022-01-05:
ah no, I used Ubuntu 20.04 with gcc 9.2.1 but in issue #900 there are some weird stuff happening with I try it out in the versions that you mentioned.

knmcguire, 2022-01-05:
I tried to use the same compiler you used (gcc 7.3, ubuntu 18.04) but the crazyflie did not default to the kalman filter without an positioning deck, so it is probably not because of your compiler. 

Did you do` make clean && make all -j8` ? with no additional flags added or in your tools/make/config.mk file if you have one?

knmcguire, 2022-01-13:
btw, if in the mean time you need to really be able to build your firmware, there is also the option of using the bitcraze toolbelt: https://github.com/bitcraze/toolbelt which builds the firmware within a docker container that we also use in our CI.  And there are some instructions in the[ installation instructions ](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/build/#compiling) to build it. You still need to do make load or use cfloader to flash the crazyflie.

`cfloader flash cf2.bin stm32-fw -w radio://0/40/2M/E7E7E7E703`

Moj-Dev, 2022-01-25:
I used "make clean && make all -j8" with no additional flag added in tools/make/config.mk file. I'm still getting "ESTKALMAN: State out of bounds, resetting".
FYI, I cloned a fresh and latest firmware. 
I flashed the crazyflie using the CF client and it works well. It means the hardware and sensors don't have any problems.

Moj-Dev, 2022-01-25:
@knmcguire Did you try with Crazyflie Bolt? Because I have a problem on Bolt and the firmware works well on Crazyflie 2.1.

Moj-Dev, 2022-01-25:
@knmcguire  Did you try with Crazyflie Bolt? Because I have this problem with Bolt and the firmware works well on Crazyflie 2.1.

knmcguire, 2022-01-25:
No I haven't but at least to my understanding that shouldn't matter as the firmware on the bolt detects it is on a bolt and configure everything properly. But maybe there are still some ifs and buts that are due to the Early Access status that hasn't been fixed yet. @tobbeanton, do you have thoughts on this?

Moj-Dev, 2022-01-27:
I'm still having problem with flashing the firmware on Bolt. I appreciate your help and support on this issue.

Moj-Dev, 2022-01-28:
I cloned the branch by tag of "2021.01" and it works well. Do you know what is the problem of latest version that produces Kalman error on Bolt?

tobbeanton, 2022-01-28:
@Moj-Dev I tried reproducing it but can't on either 2021.06 or 2022.01_rc2. Could you provide the cfclient console output after you connected and get the kalman reset?

Moj-Dev, 2022-01-28:
Here are the console's outputs:

SYS: ----------------------------
SYS: Crazyflie Bolt is up and running!
SYS: Build 0:fc1cbd422914 (2022.01) CLEAN
SYS: I am 0x20373347344D500E0038003A and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 2 deck(s) found
DECK_CORE: Calling INIT on driver bcFlow2 for deck 0
ZR2: Z-down sensor [OK]
PMW: Motion chip id: 0x49:0xB6
DECK_CORE: Calling INIT on driver bcMultiranger for deck 1
IMU: BMI088: Using SPI interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushless motor driver
SYS: About to run tests in system.c.
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
MR: Init front sensor [OK]
MR: Init back sensor [OK]
MR: Init up sensor [OK]
MR: Init left sensor [OK]
MR: Init right sensor [OK]
DECK_CORE: Deck 1 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 16360 bytes
STAB: Ready to fly.
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: State out of bounds, resetting

tobbeanton, 2022-01-31:
@Moj-Dev If the Flow deck is mounted this will switch the estimator to Kalman as per design. The problem is probably that the Flow deck is not able to track and "runs" away. Is it too close to the ground maybe?

Moj-Dev, 2022-01-31:
I don't think so!
I have enough experience working with CF and never seen this before. Although, when I flash the CF Bolt using CF client, it works well or when I use the old version of firmware like 2021.01, it works well.

BTW, I removed the flowdeck and flashed the CF without any decks. I don't get "ESTKALMAN: State out of bounds, resetting", but IMU doesn't work well. Here is the screenshot from my CF client while the crazyflie bolt is level.
[image]

tobbeanton, 2022-02-01:
I think we need to rewind a bit because I'm not sure I get the original problem any more.
1. Is there a problem with the Kalman estimator being activated even though there is no position deck mounted? The Flow deck is considered a position deck.
2. If it is mounted are you sure it is not related to the tracking output from the flow deck. A "run away"? 
3. Is this related to you own build and modified firmware? 
4. If it is a released FW what version are you experiencing the problem in?
5. If complementary filter is not level it could be the trim configuration found in Connect->Configure 2.X

knmcguire, 2022-03-07:
I'll close this due to inactivity, but please let us know @Moj-Dev if this is still a problem, and reopen it if it still is.

closed 2022-03-07 by knmcguire