bitcraze/crazyflie-firmware #915: SD card logs only timestamps for fixedfrequency

dominiknatter, 2022-01-14:
Hi everyone,

I'm currently trying to ramp up logging via the SD deck. I get to the point where log files are created on the SD card using cfclient to start and end the logging. However, when analyzing the log files using your tools (cfusdlog.py) it seems like only timestamps are being logged. We use a slightly individually adapted firmware, but those changes should mainly regard the UWB code. Here is the config.txt file, the console output, and the resulting log file:

config.txt
`1     # version
512   # buffer size in bytes
sd_log   # file name
0     # enable on startup (0/1)
on:fixedFrequency
1000     # frequency
1     # mode (0: disabled, 1: synchronous stabilizer, 2: asynchronous)
# IMU. optional
acc.x
acc.y
acc.z
gyro.x
gyro.y
gyro.z
`

console output (... replaces some irrelevant output from our own firmware):
`SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 399:29116e6eee76 (2020.06 +399) CLEAN
SYS: I am 0x20303743575350170023001F and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcUSD for deck 0
uSD: mount SD-Card [OK].
IMU: BMI088: Using I2C interface.
uSD: wait for sensors
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
STAB: Wait for sensor calibration...
SYS: Free heap: 18808 bytes
...
STAB: Ready to fly.
uSD: Config read [OK].
uSD: malloc buffer 512 bytes uSD: [OK].
uSD: Logging to: sd_log08
uSD: Wrote 1315898 B to: sd_log08 (131587 of 133000 events)
`

log file (I had to add .txt in order to upload it on GitHub, but it's a binary without file extension):
[sd_log08.txt](https://github.com/bitcraze/crazyflie-firmware/files/7872025/sd_log08.txt)

Interestingly, logging works when I don't rely on the fixedfrequency part of the config.txt file. I.e., if I use your standard example [config_kalman.txt](https://github.com/bitcraze/crazyflie-firmware/blob/master/tools/usdlog/config_kalman.txt). This will for now be my current way forward, but it'd be interesting to find the problem nonetheless.

Best,
Dominik

knmcguire, 2022-01-26:
Just for context, the SD card issue is being discussed on[ this forum thread. ](https://forum.bitcraze.io/viewtopic.php?t=5221)

tobbeanton, 2022-01-26:
This issue turned out to be the "# IMU" line comment which the parser interprets as there are no variables to log. Removing that line fixes the fixed frequency logging.

I will mark this as an enhancement and either the parser should be improved or the documentation should be clearer.

knmcguire, 2022-03-07:
I would vote parsing myself, which would require an addition to the driver in the `f_gets_without_comments `function: https://github.com/bitcraze/crazyflie-firmware/blob/633cd34ea5f88f92fe55b7f16d83cd99f967f04f/src/deck/drivers/src/usddeck.c#L404

But let's discuss it during the triage meeting.

krichardsson, 2022-05-03:
Fixed by #1004

closed 2022-05-03 by krichardsson