bitcraze/crazyflie-firmware #918: Does Crossing Beam method work stable and accurate with only one base estation?

Moj-Dev, 2022-01-21:
I'd like to use Lighthouse positioning system in my application where a human as a subject stands in the center of the testbed and a Crazyflie flies around him/her. It means if we use two base stations there are some spots without line of sight (LOS) to one of the stations. So it is necessary for the positioning deck on Crazyflie to be in LOS of both stations? Also according to the Bitcraze developers, the crossing beam method currently is more stable compared to the EKF method and in most cases on forums and websites,  Bitcraze developers mentioned that the LH works well with only one station. Did you mean using with EKF? Because the Crossing Beam requires two base stations to work. I'm a bit confused about which method is more robust, accurate and which one can work with one station. I appreciate it if you could make it clear for me.

krichardsson, 2022-01-24:
Hi!

Crossing beam requires data from two base stations as it is based on the point where the beams from the two base stations meet. I think you should use the "raw sweeps" method (it is the default setting) as it works with only one base station (two is better though).

Both methods use the EKF but crossing beam does more pre-processing and passes a full position solution to the EKF while the raw sweeps method sends sweep angles to the EKF for processing.

Which one is more robust or better? They have different properties and are robust in different ways. With full reception of two base stations, the accuracy is probably similar.

You can read more about the two methods here https://www.bitcraze.io/documentation/repository/crazyflie-firmware/2021.06/functional-areas/lighthouse/positioning_methods/

krichardsson, 2022-02-01:
@Moj-Dev Are you happy with the answer? Can we close the issue?

closed 2022-02-25 by krichardsson