bitcraze/crazyflie-firmware #923: Add USB message to reboot to DFU bootloader

ataffanel, 2022-01-27:
A need has been expressed on the forum to be able to easily program the STM32 using USB. Currently DFU programming is implemented as a failsafe functionality and so is pretty hard to access. 

If there was a way to reboot in dfu mode from USB, we could implement a ```make flash-dfu``` make target that restarts the Crazyflie in DFU mode, flash it and restart it in firmware mode. This would make working with the Crazyflie without radio a lot easier and could also help regular with-radio Crazyflie development since flashing via USB should be at least twice as fast as flashing via the radio.

I described in the forum a way to implement this functionality: https://forum.bitcraze.io/viewtopic.php?p=21533#p21533

C-Glick, 2022-01-28:
Hi ataffanel, I'd be interested in looking into this further (I was the one who recently mentioned this topic on the forum thread). 

I'm fairly new to the Crazyflie firmware so I'm trying to find my way as I go and could use some guidance. The code you linked for rebooting the CF into DFU mode seems complete (https://github.com/bitcraze/lps-node-firmware/pull/19/files) just need to run those functions at the right location.

I am trying to understand the USB setup request now.
https://github.com/bitcraze/crazyflie-firmware/blob/61f3506cce6bfbed21808676caf38d807f823c45/src/hal/src/usb.c#L167

 Is `usbd_cf_Setup()` only called when a USB cable is initially connected to the CF? It looks like right now all it does is set the CRTP link to the USB link when the index is 0x01.

Ideally, the CF would already be connected via USB and we would then send a signal (through CRTP?) over USB to initiate the reboot. Is this possible using a setup request?

C-Glick, 2022-02-09:
@ataffanel I've done some work on this but have been having issues and would like some guidance. 
I've got the crazyflie 2.0 successfully responding to a USB request but the code to jump to the DFU mode doesn't appear to be working. It doesn't seem the HAL functions used in the example given are available in the stock firmware, so I converted them to what appears to be equivalent functions. When responding to the USB request, the crazyflie reboots the firmware into a failed self-test state (red led flashing 5 times) if I attempt to jump to DFU mode again it completely locks up and I need to remove power to restore functionality. 

From what I can tell, I need to set the Main stack pointer to whatever the value is at the start of system memory, address 0x1FFF0000 for the STM32F405 found from [ST AN2606 section 71](https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf). I don't have anyway to view what this value is, but from the example code, they remap the memory address to system flash, then set the main stack pointer to 0x00000000. Once that is set you jump to the memory address 4 bytes after the start of system memory, so after the remap that should be 0x00000004.

I found this tutorial helpful for what is needed to jump to DFU mode: https://www.youtube.com/watch?v=cvKC-4tCRgw

This is the code I have so far:

https://git.ece.iastate.edu/danc/MicroCART/-/blob/dfu_flash/crazyflie_software/crazyflie-firmware-2021.06/src/hal/src/usb.c#L182

```c
} else if(command == 0x02){
    DEBUG_PRINT("default stack pointer address: %p", 0x1FFFEC00);
    //transition to DFU bootloader mode

    //shutdown any tasks running
    RCC_DeInit();
    SysTick->CTRL = 0;  //reset the Systick timer
    SysTick->LOAD = 0;
    SysTick->VAL  = 0;

    __set_PRIMASK(1);   //disable interrupts

    SYSCFG_MemoryRemapConfig(SYSCFG_MemoryRemap_SystemFlash);

    //set main stack pointer to its default value
    __set_MSP(*((uint32_t*) 0x00000000));
    //__ASM volatile ("movs r3, #0\nldr r3, [r3, #0]\nMSR msp, r3\n" : : : "r3", "sp");

    //jump to boot loader (start of system memory plus 4)
    ((void (*)(void)) *((uint32_t*) 0x00000004))();

    while(1);

  }
```

C-Glick, 2022-02-24:
Merged in commit 6c33f7eae10e539c9c1affc015776651aeb37e2e

knmcguire, 2022-03-07:
I guess we can close this right? The one thing that is perhaps missing, is double checking if the[ DFU documentatio](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/development/dfu.md)n is still valid and if we can add anything to let users know that this is possible (still for only development purposes)

C-Glick, 2022-03-10:
Hi @ruimscarvalho98 thanks for the feedback! Glad this feature is useful for you!

That would make sense if the make file changes to a new directory, the python script wouldn't be findable. Do other script calls in the make file also have this issue? Does your new srcteee variable find the absolute file path for the root of the project at the beginning of the make file?

For the .dfu file, the reason I am using that instead of the .bin file is because there is no risk of overwriting the boot loader with the .dfu file, it only overwrites the firmware automatically. With the .bin file we need to specify the address to write it to. Using the .bin file would work fine but it felt like an unnecessary risk at the time. 

I thought I remember the .dfu file being built by default, although there have been a lot of changes to the make file recently so that could have changed.

C-Glick, 2022-03-10:
> I guess we can close this right? The one thing that is perhaps missing, is double checking if the[ DFU [...]

I can write up some new documentation next week. 👍🏻

C-Glick, 2022-03-10:
Oh in that case yeah we should definitely be using the srctree variable. My bad!

I will also put a note in the documentation I will be writing next week about how to compile the .dfu file.

knmcguire, 2022-03-11:
Thanks for handling this @C-Glick ! 

And indeed, let us know if there is a PR by referring to this issue as well so that they are linked. Once we got the makefile and/or documentation have been fixed, we can close this issue

C-Glick, 2022-03-26:
Apologies I haven't been able to get around to this yet. Some big deadlines came up and I've been busy. Will hopefully get this resolved in the next few weeks. Next week will also be pretty busy for me but should hopefully have some free time after that.

knmcguire, 2022-03-29:
No problem! give us an headsup ones you have time to work on it again :smile:

C-Glick, 2022-05-09:
My doc changes along with a couple other improvements have been merged into master. If there is nothing else to address this issue can probably be closed.

krichardsson, 2022-05-10:
Fixed in #1040

closed 2022-05-10 by krichardsson