bitcraze/crazyflie-firmware #933: assert error in param_logic in app layer setup

dominiknatter, 2022-02-03:
Hey everyone,

we are currently using the crazyflie-firmware as a submodule in our work (based on your app layer functionality). So far we have relied on the firmware version 2021.06 and all worked well. Yesterday I updated to the newest firmware. Now, when starting the drone it restarts and then blinks with the 5-time-red LED error code. Looking at the following console output, there seems to be an assert() error.

SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 24:c2790c0d9ed6 (2022.01 +24) MODIFIED
SYS: I am 0x20303743575350170038004F and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 3 deck(s) found
DECK_CORE: Calling INIT on driver bcMultiranger for deck 0
DECK_CORE: Calling INIT on driver bcDWM1000 for deck 1
DECK_CORE: Calling INIT on driver bcFlow2 for deck 2
ZR2: Z-down sensor [OK]
PMW: Motion chip id: 0x49:0xB6
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
MR: Init front sensor [OK]
MR: Init back sensor [OK]
MR: Init up sensor [OK]
MR: Init left sensor [OK]
MR: Init right sensor [OK]
DECK_CORE: Deck 0 test [OK].
DECK_CORE: Deck 1 test [OK].
DECK_CORE: Deck 2 test [OK].
SYS: The system resumed after a failed assert [WARNING]
SYS: Assert failed at crazyflie-firmware/src/modules/src/param_logic.c:576
SYS: cfAssertNormalStart [FAIL]

[The mentioned line](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/param_logic.c#L576) in the code tests a float parameter with PARAM_VARID_IS_VALID() and that apparently fails. I've tried to set a DEBUG_PRINT() at line 575 to figure out which parameter it is, but unfortunately I never got any console output for this print (not sure what I'm doing wrong there).

We have added some minor changes to the UWB part of the firmware. In those changes, we have added one new parameter. However, if I comment out that section the error still remains. So this should not cause the problem.
Looking at the console output I don't think the remaining part of our app layer has been started yet, so I would not see how any param calls from there can lead to this assert either.

I see there has been a lot of changes due to the [persistent param introduction](https://github.com/bitcraze/crazyflie-firmware/pull/892), maybe something in there messes with us?
Or is there a possibility that I should not focus on parameters because that is just what is visible in the console output but the underlying problem is something different?

Sorry if this report is not very detailed, but maybe you can still point me in the right directions for fixing this problem. We'd like to rely on the newest firmware to integrate new features like the SD card.

Best,
Dominik

dominiknatter, 2022-02-03:
ahh well apparently our app code was already being called. In there we had a call
`paramSetFloat(paramGetVarId("posCtlPid","rpLimit"),5.0f);`
This parameter does not exist anymore as it seems [since November 2021](https://github.com/bitcraze/crazyflie-firmware/commit/e0aea900143a57a6f9f3ad7a3be8280379111a5b#diff-79636f5296852f7bfb39dad077633dceed42aaef69c5c8d1d1ef588d54849e9d). Sorry for the false alarm.

closed 2022-02-03 by dominiknatter

knmcguire, 2022-02-04:
fwew, luckily it wasn't an error in the current firmware. We did do a lot of changes to the parameter framework so it seems plausible that here can be errors somewhere that we missed. 

Happy that you found your issue!